isaac-sim/IsaacLab #1: [Bug Report] Program aborted the execution due to unhandled error when runing API Demos 1 and 2

Toni-SM, 2023-01-18:
### Describe the bug

Demos 1 and 2 in [API Demos](https://isaac-orbit.github.io/orbit/source/setup/sample.html#api-demos) abort the execution due to unhandled error

### Steps to reproduce

Run demos 1 and 2 in [API Demos](https://isaac-orbit.github.io/orbit/source/setup/sample.html#api-demos) 

- Spawn different quadrupeds, visualize feet markers, and make robots stand using position commands:
  
  ```bash
  ./orbit.sh -p source/standalone/demo/play_quadrupeds.py
  ```

  [kit_20230118_201202.log](https://github.com/NVIDIA-Omniverse/orbit/files/10449933/kit_20230118_201202.log)
  
  [image]

- Spawn multiple Franka arms and apply random position commands:

  ```bash
  ./orbit.sh -p source/standalone/demo/play_arms.py --robot franka_panda
  ```
  [kit_20230118_201501.log](https://github.com/NVIDIA-Omniverse/orbit/files/10449942/kit_20230118_201501.log)

  [image]

### System Info

Describe the characteristic of your environment:

- Commit: dd24040b115fbaa705a83fee4eac9afaa1481be0
- Isaac Sim Version: 2022.2.0
- OS: Ubuntu 20.04
- GPU: RTX 3080 
- CUDA: 11.8
- GPU Driver: 520.61.05

### Additional context

- Demo 3 runs well

  [image]

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2023-01-25:
Just to understand what could be happening, could you also check if the issue happens when running Isaac Sim example on multiple frankas. It does a similar set of operations.

```bash
cd ${ISAACSIM_PATH}
./python.sh standalone_examples/api/omni.isaac.core/add_frankas.py
```

Toni-SM, 2023-01-25:
There is no error running the Isaac Sim example

[image]

Demos 1 and 2 still crashing

ADebor, 2023-03-14:
Hi @Toni-SM, @Mayankm96 ,

I don't know if you figured out what is happening, but I just wanted to let you know I'm experiencing the exact same problem with demos 1 and 2, as well as with the teleoperation script located at `source/standalone/environments/teleoperation/teleop_se3_agent.py`, as you can see below.

[image]

Just like @Toni-SM, there is no error running the Isaac Sim example.

My environment:
Commit: [`7fe10e1`](https://github.com/NVIDIA-Omniverse/Orbit/commit/7fe10e1c87b2d00c701e86b90639f898f83641ea)
Isaac Sim Version: 2022.2.0
OS: Ubuntu 20.04
GPU: RTX 3060
CUDA: 12.0
GPU Driver: 525.85.05

Mayankm96, 2023-03-16:
Hi @ADebor ,

Thanks for bringing this up again. We weren't able to find the issue yet but from what I discussed with Toni, he was having this issue when running the scripts on a laptop (they worked fine on his desktop with a similar configuration).

For you, is this happening on a laptop setup as well? Do other scripts also not work (example: `play_arms.py`)?

ADebor, 2023-03-16:
Hi @Mayankm96, 

Thanks for your quick answer. 

Yes, the problem does happen when running the scripts on my laptop (Thinkpad X1 Extreme Gen5).  As for the scripts, I encounter the problem running the two first demos `play_quadrupeds.py`, `play_arms.py`, and the teleop one I mentioned in my previous comment. `play_ik_control.py` works though.

Mayankm96, 2023-03-24:
Hi @Toni-SM and @ADebor ,

[Isaac Sim 2022.2.1](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/release_notes.html) has been released recently. Can it be checked if the issue still persists with it?

Thanks a lot!

Toni-SM, 2023-03-24:
Hi @Mayankm96 

Demo 1 and 2 still fail
Demo 3 and 4 run fine

[demo2-kit_20230324_105000.log](https://github.com/NVIDIA-Omniverse/Orbit/files/11060648/demo2-kit_20230324_105000.log)
[demo1-kit_20230324_104833.log](https://github.com/NVIDIA-Omniverse/Orbit/files/11060649/demo1-kit_20230324_104833.log)

```txt
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 520.61.05    Driver Version: 520.61.05    CUDA Version: 11.8     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  On   | 00000000:01:00.0 Off |                  N/A |
| N/A   47C    P8    18W /  N/A |     12MiB / 16384MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1574      G   /usr/lib/xorg/Xorg                  4MiB |
|    0   N/A  N/A      2484      G   /usr/lib/xorg/Xorg                  4MiB |
+-----------------------------------------------------------------------------+

```

Mayankm96, 2023-03-28:
The issue with `orbit.sh --extra` should be fixed now #41 . For the demos, I have requested for a laptop with a GPU to see what could be the issue here.

zhuyijie88, 2023-03-30:
@Mayankm96 , it still has the same fault with branch [fix/device-in-demos].

My environment:
Commit: b7263b0
Isaac Sim Version: 2022.2.1
OS: Ubuntu 20.04
GPU: RTX 3080
CUDA: 11.3
GPU Driver: 525.60.11

zhuyijie88, 2023-03-31:
Adding ```device='cuda'``` solves the problem. For example, in Line 91 of play_quadrupeds.py, change the code to:
```
sim = SimulationContext(stage_units_in_meters=1.0, physics_dt=0.005, rendering_dt=0.005, backend="torch", device='cuda')
```

ADebor, 2023-03-31:
Hi there,

Sorry for the late response. 

- Demos 1 and 2 work fine on my end with the new Isaac Sim release, using the same code as before. Demos 3 and 4 still work as well. I've just git pulled the repo's main branch and these 4 demos still work just fine. No need to add `device='cuda'` on my end.

- `orbit.sh --extra` does not fail anymore, as expected. 

- The teleop example (`source/standalone/environments/teleoperation/teleop_se3_agent.py`) still fails though. When launched using the `--cpu` option or not, the script runs, but I'm not able to control the robot using the keyboard. Or maybe I don't use this example correctly: I simply launch it in the terminal and press keys on my keyboard, making them printed in this same terminal, but not moving the robot in the simulator. When pressing keys in the simulator, 'K' does toggle the gripper (which is expected), but pressing other keys simply selects different modes in the left panel of the simulator.

Toni-SM, 2023-03-31:
> Adding `device='cuda'` solves the problem.

This change works. However, robots do nothing!

ADebor, 2023-03-31:
Hi @Toni-SM ,

Did you wait for several "Resetting robots state..."? The robots do move on my side, but only to reach a random pose once per reset. They do not move randomly at all steps.

Toni-SM, 2023-03-31:
Hi @ADebor 

I have run the example for quite some time but nothing.
The robots do not go to a random pose at each reset nor do they open or close the gripper, as they do when I run the code on a workstation.

Mayankm96, 2023-03-31:
@Toni-SM, on the UI it says to enable scene graph instancing? Can you select that to enable it?

Toni-SM, 2023-03-31:
Hi @Mayankm96 

Enabling the scene graph instancing in UI does not do anything

[image]

kumar-sanjeeev, 2023-06-26:
Hello @Toni-SM and @Mayankm96,

Now I am able to run the robot using the `source/standalone/demo/play_arms.py` script by specifying `device="cuda"`. I made the following changes in the source code, referring to `source/standalone/demo/play_cloner.py`.

#### Changes
```
from omni.isaac.core.utils.carb import set_carb_setting

def main():
    """Spawns a single arm manipulator and applies random joint commands."""

    # Load kit helper
    sim = SimulationContext(physics_dt=0.01, rendering_dt=0.01, backend="torch", device="cuda")
    # Set main camera
    set_camera_view([2.5, 2.5, 2.5], [0.0, 0.0, 0.0])
    
    ############# changes
    # Enable fltacache which avoids passing data over to USD structure
    # this speed up the read-write operation of GPU buffers
    if sim.get_physics_context().use_gpu_pipeline:
        sim.get_physics_context().enable_flatcache(True)
    # Enable hydra scene-graph instancing
    # this is needed to visualize the scene when flatcache is enabled
    set_carb_setting(sim._settings, "/persistent/omnihydra/useSceneGraphInstancing", True)
    ###############

```
After these changes, I no longer need to **enable scene graph instancing**, but I still need to specify the **device="cuda"**.

Mayankm96, 2023-08-23:
I finally got my hands on a machine locally where we could reproduce this issue. This is happening on systems that have an Intel graphics card along with an NVIDIA RTX card. If the monitor is configured to use the Intel card, then the CPU simulation is failing while the GPU simulation works fine.

At least for the PC where we saw this issue, we just plugged the monitor to the NVIDIA graphics card and it started working fine. On laptops, I suppose you can use [`prime-select`](https://askubuntu.com/questions/1106854/use-displays-connected-to-different-graphic-cards).

Would be great to know if this fixes the issues.

Mayankm96, 2024-02-10:
Since the project has undergone large changes and no further response on this issue, I am closing it up. 

I hope the fix above does solve the problem for you. If the issue persists with the latest main, please feel free to open the issue again.

closed 2024-02-10 by Mayankm96