isaac-sim/IsaacLab #4: [Question] Visual Observations Setup?

StoneT2000, 2023-01-19:
### Question
Awesome paper and work so far, I was curious about setting up environments with visual observations. Is that possible yet? If so how do you configure that? I couldn't find anything about visual observations (for RL workflows) in the docs.

Thanks

Mayankm96, 2023-01-19:
Thank you for your question!

We are in the process of extending the observation and rewards manager to also handle different sensors in the scene. This would allow users to define sensor configurations and obtain data from them in their observations.

However, currently, Isaac Sim does not support parallelized cameras. So the expected visual throughput would be slow. As discussed in #3, we will prioritize extending the current camera class to have a `CameraView` class that at least gets the data from different cameras. However, this will be an experimental class since we do not have a clear definition of what parallel camera APIs will look like from the Omniverse side.

I am going to leave this issue open for discussion.

Mayankm96, 2023-02-24:
Sharing this update here for what's in store in the coming releases of Isaac Sim:

> https://www.youtube.com/watch?v=BVhJ-6D6wcs&ab_channel=ErwinCoumans

nikepupu, 2023-02-26:
@Mayankm96 I went through the code for the link. It seems he is using a separate render which has nothing to do with omniverse to render the cartpole urdf only.