isaac-sim/IsaacLab #18: [Question] Accessing Assets

peterdavidfagan, 2023-02-08:
### Question

While developing a Docker container setup for Orbit I have encountered issues in running the demos. In particular I am seeing warnings related to accessing assets and an error generated by `set_articulation_properties` in the `kit.py` file. It seems like the path provided for the articulation primitive is incorrect (https://github.com/peterdavidfagan/Orbit/blob/260d41168d04b699113e9e3b7af38f98dd1493e4/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/kit.py#L271).  

Posting here for visibility in case there is a quick fix for this.

```
orbit_1      | [145.454s] Simulation App Startup Complete
orbit_1      | 2023-02-08 15:54:02 [147,443ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2830 of /buildAgent/work/ca6c508eae419cf8/USD/pxr/usd/usd/stage.cpp -- Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.1/Isaac/Props/Mounts/SeattleLabTable/table_instanceable.usd@ for reference on prim @anon:0x254cb410:World0.usd@,@anon:0x22ea3a60:World0-session.usda@</World/Table_1>. (recomposing stage on stage @anon:0x254cb410:World0.usd@ <0x2509eb30>)
orbit_1      | 
orbit_1      | 2023-02-08 15:54:02 [147,447ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2830 of /buildAgent/work/ca6c508eae419cf8/USD/pxr/usd/usd/stage.cpp -- Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.1/Isaac/Props/Mounts/SeattleLabTable/table_instanceable.usd@ for reference on prim @anon:0x254cb410:World0.usd@,@anon:0x22ea3a60:World0-session.usda@</World/Table_2>. (recomposing stage on stage @anon:0x254cb410:World0.usd@ <0x2509eb30>)
orbit_1      | 
orbit_1      | 2023-02-08 15:54:02 [147,645ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2830 of /buildAgent/work/ca6c508eae419cf8/USD/pxr/usd/usd/stage.cpp -- Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2022.1/Isaac/Samples/Orbit/Robots/FrankaEmika/panda_instanceable.usd@ for reference on prim @anon:0x254cb410:World0.usd@,@anon:0x22ea3a60:World0-session.usda@</World/Robot_1>. (recomposing stage on stage @anon:0x254cb410:World0.usd@ <0x2509eb30>)
orbit_1      | 
orbit_1      | Traceback (most recent call last):
orbit_1      |   File "./source/standalone/demo/play_arms.py", line 159, in <module>
orbit_1      |     main()
orbit_1      |   File "./source/standalone/demo/play_arms.py", line 94, in main
orbit_1      |     robot.spawn("/World/Robot_1", translation=(0.0, -1.0, 0.0))
orbit_1      |   File "/isaac-sim/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/robot_base.py", line 150, in spawn
orbit_1      |     solver_velocity_iteration_count=self.cfg.articulation_props.solver_velocity_iteration_count,
orbit_1      |   File "/isaac-sim/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/kit.py", line 274, in set_articulation_properties
orbit_1      |     raise ValueError(f"No articulation schema present for prim '{prim_path}'.")
orbit_1      | ValueError: No articulation schema present for prim '/World/Robot_1'.
orbit_1      | /isaac-sim/python.sh: line 40:    23 Segmentation fault      (core dumped) $python_exe "$@" $args
orbit_1      | There was an error running python
docker_orbit_1 exited with code 1

```

peterdavidfagan, 2023-02-16:
I am starting to revisit this, I have in the meantime been working with the workstation installation. 

It is likely the case that this error is due to the Isaac-sim version that is being used. Some of the required assets are not seen on my local Nucleus for Isaac 2022.1 however they are for 2022.2. 

@sabdulajees are there plans to release and official dockerfile for 2022.2.0 within the following [repo](https://github.com/NVIDIA-Omniverse/IsaacSim-dockerfiles)?

Mayankm96, 2023-05-26:
Closing the issue due to no activity. We will look into the docker container in the coming sprint in June. 

Please re-open if still unresolved or if you have any additional comments.

closed 2023-05-26 by Mayankm96

Jianlinlynn, 2023-11-29:
HI @peterdavidfagan , I would like to know have you solved the problem by updating? I met almost the same one for my self-designed robot (https://github.com/NVIDIA-Omniverse/Orbit/issues/152). 

Do you have any suggestions for that? Thanks a lot for any reply!

peterdavidfagan, 2023-11-29:
Hi @Jianlinlynn, 

From what I recall, I could only solve the problem by upgrading to the latest version. It's been a while since I worked with this platform but to the best of my knowledge this is how I fixed the problem for me. 

Cool project by the way.

Jianlinlynn, 2023-11-29:
Dear @peterdavidfagan , thank you so much for your reply!