isaac-sim/IsaacLab #34: Slight error in lift_env reward manager

adamwhats, 2023-03-07:
https://github.com/NVIDIA-Omniverse/Orbit/blob/d75e7ff63bf08b558266d2eff42fd4db15d29d6b/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/manipulation/lift/lift_env.py#L399-L401

Looks like this missing the exp-kernel, assuming it should be something along the lines of:
```
  def reaching_object_position_exp(self, env: LiftEnv, sigma: float):
      """Penalize end-effector tracking position error using exp-kernel."""
      error = torch.sum(torch.square(env.robot.data.ee_state_w[:, 0:3] - env.object.data.root_pos_w), dim=1)
      return torch.exp(-error / sigma)
```

Mayankm96, 2023-03-07:
Hi, yes this was an issue. 

I made a PR yesterday #33 that tunes the rewards for the lift environment to work with rsl_rl library. Haven't merged it yet. Need some time to tune it for rl_games and sb3 too.

adamwhats, 2023-03-07:
Great news, look forward to the merge!

commit 64f810fa (2023-10-25): Add spawner functionality to allow direct loading of OBJ/STL/FBX assets (#138)

# Description

This PR enables users to convert meshes from OBJ, STL or FBX format to
USD with a few options:
- Asset instanceable
- Mass properties
- Rigid body properties
- Collision properties - including mesh approximation method

The layer hierarchy is as follows;
```
/World/
     [MESH_OBJECT_NAME] - Xform
         [INPUT_MESH_NAME]_xform - Xform
                /Looks/ - Material
                [MESH] - Mesh
```

Fixes #34. 

## Type of change

- New feature (non-breaking change which adds functionality)

# [... cut]
closed 2023-03-20 by Mayankm96