isaac-sim/IsaacLab #37: [Question] How to assign dof/joint indices?

shijiyuanaa, 2023-03-09:
I'm trying to use `LeggedMobileManipulator` to combine a quadruped and a robot arm. I use 'convert_urdf.py' to convert a URDF to USD. However, the joint indices were not as  [base_dof_nums, arm_dof_nums, tool_dof_nums(which is 0 in my case)] after the robot init. (The dof indices were initialized in line 212 of `articulation_view.py`.) So there may be some problems when I call robot.get_default_dof_state() or robot.apply_action(), etc.

My question is how to make sure the joint order is like [base, arm, tool] when importing the USD file? Thanks a lot!

Mayankm96, 2023-03-20:
Hi, thanks for pointing this out!

I tried the same for our ALMA (arm-on-ANYmal) robot last week and also faced the same issue. The parsing of joints in Isaac Sim is done in a breadth-first order instead of depth-first, which means the legs and arm joints are arranged differently to what the MobileManipulator class expects (base joints and arm joints in chunks).

We can't change the way PhysX parses the joints. This convention is defined by USDPhysics. However, a possible solution can be that the users specify through the configuration object, which joints belong to different parts of a heterogeneous robot.

The interface would then become:
1. User sets and gets the joints at their pre-specified order
2. The class converts them into simulation order

Would this be something that makes sense?

shijiyuanaa, 2023-03-20:
> Hi, thanks for pointing this out!
> 
> I tried the same for our ALMA (arm-on-ANYmal) robot last week and also faced the same issue. The parsing of joints in Isaac Sim is done in a breadth-first order instead of depth-first, which means the legs and arm joints are arranged differently to what the MobileManipulator class expects (base joints and arm joints in chunks).
> 
> We can't change the way PhysX parses the joints. This convention is defined by USDPhysics. However, a possible solution can be that the users specify through the configuration object, which joints belong to different parts of a heterogeneous robot. This way the inner workings of the joints handled by the physics engine will remain hidden from the user and they only "see" joints in the order they specify. If they don't specify an order then the default ordering is used.
> 
> The interface would then become:
> 
> 1. User sets and gets the joints at their pre-specified order
> 2. The class converts them into simulation order
> 
> Would this be something that makes sense?

Thank you for your reply! 
I tried to change the parent link of the robot arm's first joint in my urdf and it works 😸 . But I hope there be more elegant way to change joint order. Thank you anyway!

Mayankm96, 2023-03-20:
Interesting. Could you elaborate more on what you mean by:
> I tried to change the parent link of the robot arm's first joint in my urdf and it works

shijiyuanaa, 2023-03-20:
> Interesting. Could you please elaborate more on what you mean by:
> 
> > I tried to change the parent link of the robot arm's first joint in my urdf and it works

In the original urdf file, I simply attach the arm to the `trunk` link of the quadruped, and I found that the joint order is not what I want. Then I attach the arm to `imu_link`, which is a child link of `trunk`(so that the arm's fixed joint and the thigh joints are in the same hierarchy , I guess ), then I got the correct joint order.

Murkey8895, 2023-04-04:
> In the original urdf file, I simply attach the arm to the `trunk` link of the quadruped, and I found that the joint order is not what I want. Then I attach the arm to `imu_link`, which is a child [...]

Hi, I'm also trying to combine a quadruped and a robot arm. Following your idea, I attach the arm's first fixed joint to `imu_link`, but still get a wrong  order:
`['joint1', 'joint2', 'joint3', 'joint4', 'FL_hip_joint', 'FR_hip_joint', 'RL_hip_joint', 'RR_hip_joint', 'joint5', 'FL_thigh_joint', 'FR_thigh_joint', 'RL_thigh_joint', 'RR_thigh_joint', 'joint6', 'FL_calf_joint', 'FR_calf_joint', 'RL_calf_joint', 'RR_calf_joint', 'jointGripper']`
where the joint[1-6] are the arm's joints. Is it convenient for you to show your urdf file to help me solve this problem? Thank you in advance!

closed 2023-12-22 by Mayankm96