isaac-sim/IsaacLab #38: Training an agent with RSL-RL on Isaac-Lift-Franka-v0

sabinakaminska95, 2023-03-10:
[Screencast from 10.03.2023 14:19:13.webm](https://user-images.githubusercontent.com/84527990/224327042-2827979a-7978-4685-ad81-db7e06ba9a73.webm)
There is no interaction with the cube, which should be picked up by the robot after training.

adamwhats, 2023-03-19:
There's a [PR](https://github.com/NVIDIA-Omniverse/Orbit/pull/33) which addresses this for RSL_RL, still WIP for the others. Worked first time for me

Mayankm96, 2023-03-20:
Sorry for the late reply. I merged that branch now since the issue has been popping up for a while. 

Right now `rsl_rl` should work but haven't been able to tune `rl_games` to make it work. Probably has something to do with differing PPO implementations.

Anyway, feel free to send an MR if you are using other libraries and managed to tune it to make it work.

closed 2023-03-20 by Mayankm96