isaac-sim/IsaacLab #39: [Question] Attach Camera to a specific link

khalidA16, 2023-03-20:
I am trying to attach a camera to my robot's link using the script from [play_camera.py](https://github.com/NVIDIA-Omniverse/Orbit/blob/751a33dd5ff656d9b1028be7a91ce05946747666/source/standalone/demo/play_camera.py).
I initialized my robot as:
`robot = my_world.scene.add(PR2(prim_path="/World/PR2", name="pr2"))`
and spawned the camera as:
`camera.spawn("/World/PR2/wide_stereo_r_stereo_camera_frame")` where the prim_path is the link where I would like to attach the camera.
Instead of attaching to the provided prim_path, it is spawned at a distance from the given link ( at a different spot and height in the air than that of the link ) 
[image]

Kindly let me know if there is a way to attach the camera to a link. Thanks

Mayankm96, 2023-03-20:
Can you try the following to see if it fixes the issue:

```python
camera.spawn("/World/PR2/wide_stereo_r_stereo_camera_frame/dummy")
```

khalidA16, 2023-03-20:
Thank you for your reply. I tried your change but now the camera is created farther.
[image]

Mayankm96, 2023-03-20:
Is it possible to get the script and asset so that we can try reproducing and fixing it from our end?

khalidA16, 2023-03-20:
> Is it possible to get the script and asset so that we can try reproducing and fixing it from our end?

It turned out that `camera.set_world_pose_ros()` was being called in the script which was messing up with the position.

I have another query, the main viewport in the gui displays the new camera created by default. If I change it back to “Perspective” from the gui, `camera.data.output` returns data from the perspective camera, and not from the camera on the robot anymore. 
Do you have a suggestion in mind for this ? Thanks

Mayankm96, 2023-03-21:
That's strange. The camera uses Hydra textures which don't require opening a new viewport. So by default, it should be opening the "Perspective" camera.

I ran the `play_camera.py` again and it shows the default camera as the perspective camera. I switched the viewport to use the created camera and then back to "Perspective". The behavior remains unchanged for me.

Mayankm96, 2023-03-21:
Ah yes. The switch from what is now called "legacy" viewport to hydra textures happened in Kit 104 which is what the IsaacSim 2022.2 release is using.

I would recommend using the latest Isaac Sim 2022.2.1 release since there are many bug fixes and improvements that have come in.

closed 2023-03-22 by Mayankm96