isaac-sim/IsaacLab #47: [Bug Report] After Isaac-Velocity-Anymal-C-v0 is trained baesd on rsl-rl,  it cannot walk.

zhuyijie88, 2023-04-03:
### Describe the bug

After Isaac-Velocity-Anymal-C-v0 is trained 1000 steps baesd on rsl-rl,  it cannot walk, but float in the air.

### Steps to reproduce

on bea049b commit,
```
./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Anymal-C-v0 --headless
./orbit.sh -p source/standalone/workflows/rsl_rl/play.py --task Isaac-Velocity-Anymal-C-v0 --num_envs 1 --checkpoint /PATH/TO/model_1000.pth
```
### System Info

Describe the characteristic of your environment:

- Commit: bea049b
- Isaac Sim Version:  2022.2.1
- OS: Ubuntu 20.04
- GPU: RTX 3080
- CUDA: 11.3
- GPU Driver: 525.60.11

### Additional context

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2023-04-07:
Hi @zhuyijie88 ,

Thank you for pointing this out. Yes, there seems to be some issue that popped up with 2022.2.1 release. Somehow even when applying zero actions, the robot appears to be floating. I am checking the issue out. Will update here as soon as I find a fix.

romesco, 2023-04-15:
Take a look here:

https://github.com/NVIDIA-Omniverse/Orbit/blob/6d855682eb7cafa4c8510ba6e33fb3a4955f2131/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/locomotion/velocity/velocity_cfg.py#L156

You might want to play with the `base_height_fall` settings.

Mayankm96, 2023-12-19:
Closing this MR as we have this working now on our devel branch.

closed 2023-12-19 by Mayankm96