isaac-sim/IsaacLab #53: [Question] where can I set maximum velocity?

PARKBONG, 2023-04-10:
### Question

Hi this is my code. I modified [frank.py](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/config/franka.py), and created my own code. 

`
"""
# from omni.isaac.orbit.actuators.config.franka import PANDA_HAND_MIMIC_GROUP_CFG
from omni.isaac.orbit.actuators.group import ActuatorGroupCfg, GripperActuatorGroupCfg
from omni.isaac.orbit.actuators.group.actuator_group_cfg import ActuatorControlCfg
from omni.isaac.orbit.actuators.model import ImplicitActuatorCfg
# from omni.isaac.orbit.utils.assets import ISAAC_ORBIT_NUCLEUS_DIR
from omni.isaac.orbit.robots.single_arm import SingleArmManipulatorCfg

# _FRANKA_PANDA_ARM_INSTANCEABLE_USD = f"{ISAAC_ORBIT_NUCLEUS_DIR}/Robots/FrankaEmika/panda_instanceable.usd"
# _ROBOTIQ_WRIST_INSTANCEABLE_USD = "/home/bong/.local/share/ov/pkg/isaac_sim-2022.2.1/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/bong/robotiq_wrist.usd"
_ROBOTIQ_WRIST_INSTANCEABLE_USD = "/home/bong/.local/share/ov/pkg/isaac_sim-2022.2.1/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/bong/robotiq_wrist_resized_direction.usd"
# _ROBOTIQ_WRIST_INSTANCEABLE_USD = "/home/bong/.local/share/ov/pkg/isaac_sim-2022.2.1/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/bong/robotiq_wrist_resized_direction_mass.usd"

VELOCITY_LIMIT = 0.000000000000001
TORQUE_LIMIT = 100
STIFFNESS = 400

ROBOTIQ_WRIST_WITH_ROBOTIQ_CFG = SingleArmManipulatorCfg(
    meta_info=SingleArmManipulatorCfg.MetaInfoCfg(
        usd_path=_ROBOTIQ_WRIST_INSTANCEABLE_USD,
        arm_num_dof=6,
        tool_num_dof=4,
        tool_sites_names=["bot_link_1", "bot_link_2", "top_link_1", "top_link_2"],  # xform
    ),
    init_state=SingleArmManipulatorCfg.InitialStateCfg(   # revjoint
        pos=[0.2, 0, 0.3],
        rot=[0.707, 0, 0.707, 0],
        dof_pos={
            "trans_x": 0.0,
            "trans_y": 0.0,
            "trans_z": 0.0,
            "rev_x": 0.0,
            "rev_y": 0.0,
            "rev_z": 0.0,
            "bot_joint_0_p": 0.0,
            "top_joint_0_p": 0.0,
            "bot_joint_1_p": 0.0,
            "top_joint_1_p": 0.0,
            "bot_joint_2_p": 0.0,
            "top_joint_2_p" : 0.0,
        },
        dof_vel={".*": 0.0},

    ),
    ee_info=SingleArmManipulatorCfg.EndEffectorFrameCfg(
        body_name="base", pos_offset=(0.0, 0.0, 0.15), rot_offset=(1.0, 0.0, 0.0, 0.0)  # xform / head
    ),
    rigid_props=SingleArmManipulatorCfg.RigidBodyPropertiesCfg(
        # max_depenetration_velocity=5.0,
        max_depenetration_velocity=0.0000000000001,
    ),
    collision_props=SingleArmManipulatorCfg.CollisionPropertiesCfg(
        contact_offset=0.005,
        rest_offset=0.0,
    ),
    articulation_props=SingleArmManipulatorCfg.ArticulationRootPropertiesCfg(
        enable_self_collisions=True,
    ),
    actuator_groups={
        "wrist_trans": ActuatorGroupCfg(
            dof_names=["trans_[x,y,z]"],
            model_cfg=ImplicitActuatorCfg(velocity_limit=VELOCITY_LIMIT, torque_limit=TORQUE_LIMIT),
            control_cfg=ActuatorControlCfg(
                command_types=["p_abs"],
                stiffness={".*": 400.0},
                damping={".*": 20.0},
                dof_pos_offset={
                    "trans_x": 0.0,
                    "trans_y": 0.0,
                    "trans_z": 0.0,
                },
            ),
        ),
        "wrist_rev": ActuatorGroupCfg(
            dof_names=["rev_[x-z]"],
            model_cfg=ImplicitActuatorCfg(velocity_limit=VELOCITY_LIMIT, torque_limit=TORQUE_LIMIT),
            control_cfg=ActuatorControlCfg(
                command_types=["p_abs"],
                stiffness={".*": 400.0},
                damping={".*": 20.0},
                dof_pos_offset={"rev_x": 0, "rev_y": 0, "rev_z": 0.0},
            ),
        ),
        "robotiq_hand": GripperActuatorGroupCfg(
            dof_names=["bot_joint_0_p", "top_joint_0_p", "bot_joint_1_p", "top_joint_1_p", "bot_joint_2_p", "top_joint_2_p"],
            model_cfg=ImplicitActuatorCfg(velocity_limit=VELOCITY_LIMIT, torque_limit=TORQUE_LIMIT),
            control_cfg=ActuatorControlCfg(command_types=["p_abs"], stiffness={".*": 400}, damping={".*": 20}),
            mimic_multiplier={"bot_joint_0_p": 1, "top_joint_0_p": -1, "bot_joint_1_p": 1, "top_joint_1_p": -1, "bot_joint_2_p": -1, "top_joint_2_p": 1},
            # speed=1,
            open_dof_pos=0,
            close_dof_pos=0.785398,
        )
    },
)
"""Configuration of Franka arm with Franka Hand using implicit actuator models."""
`
as you can see, I set `VELOCITY_LIMIT = 0.000000000000001` and `max_depenetration_velocity=0.0000000000001,`.
But the robot still moves so fast. 

How can I change it?

Mayankm96, 2023-04-10:
This seems like something that is more related to the physics engine (PhysX) in Isaac Sim.

The velocity limit is only enforced as a soft constraint in the solver and often leads to undesirable behaviors. Thus, as a practice, we keep this quantity high in order to not enforce a constraint. 

The effect you want is in the task space of the gripper test (i.e. how fast it moves) which I don't think can be handled directly by setting values in the joint space.

From what I know, you'll need to play with the PD gains to make the gripper open and close slowly. Alternatively, you can use the "v_abs" mode for the gripper which does something similar to [Robosuite](https://github.com/ARISE-Initiative/robosuite/blob/master/robosuite/models/grippers/robotiq_85_gripper.py#L64-L66), i.e. adjusting the gripper speed by clipping the values.

PARKBONG, 2023-04-11:
thank you for your reply!

since all of my robot joints (including 6-DoF wrist and 4-DoF gripper) move so fast, I just adjusted the action clipping range. 
thanks!

PARKBONG, 2023-04-13:
I think this is the option to directly limit some value of joint space.

[franka.py](https://github.com/NVIDIA-Omniverse/Orbit/blob/054974bfaf107afc53e8df0f980f585994f7c66f/source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/config/franka.py#L50)
```
  rigid_props=SingleArmManipulatorCfg.RigidBodyPropertiesCfg(
      # solver_position_iteration_count=16,  # bong
      # solver_velocity_iteration_count=1,  # bong
      linear_damping=10,
      angular_damping=100,
      max_angular_velocity=0.4,
      max_linear_velocity=0.5, 
      disable_gravity=False,
      max_depenetration_velocity=5.0,
  ),
```
 
 But the robot is showing much jerkiness.

closed 2023-04-13 by PARKBONG