isaac-sim/IsaacLab #60: collision error of mobile manipulator and YCB object

dengyh16code, 2023-04-13:
I have defined a mobile manipulator and import a YCB object in issac orbit.
here is the code of importing YCB object

    Obj_cfg = RigidObjectCfg()
    Obj_cfg.meta_info.usd_path = f"{ISAAC_NUCLEUS_DIR}/Props/YCB/Axis_Aligned_Physics/006_mustard_bottle.usd"
    rigid_object = RigidObject(Obj_cfg)

    obj_quat = convert_quat(tf.Rotation.from_euler("XYZ", (-90, 90, 0), degrees=True).as_quat(), to="wxyz")
    translation =(np.random.uniform(-0.1,0.1), np.random.uniform(-0.3,0), 0.8)
    rigid_object.spawn("/World/Objects/mustard_bottle",translation=translation, orientation=obj_quat)

and when I try to control the mobile manipulator to grasp the YCB object, there is no collision between the gripper and the object.
But there is collision between the robotic arm and the object.  I have check the URDF file, but there is no difference between the arm and the gripper.

the following video shows the grasp process. 
https://user-images.githubusercontent.com/41771334/231722830-79a4a5e2-c3ec-407b-b2a0-b8c5db67d322.mp4

dengyh16code, 2023-04-13:
I have build the robot with the class LeggedMobileManipulator and set collision property to the whole robot. I think there should be collision bodies on the gripper, but there is not.

Mayankm96, 2023-04-13:
These should be added by default when you import the URDF for your robot. If it is missing over there, then you'd need to add the collision API to your visual mesh. 

You can check the physics documentation about this:
* https://docs.omniverse.nvidia.com/prod_extensions/prod_extensions/ext_physics/rigid-bodies.html
* https://docs.omniverse.nvidia.com/prod_extensions/prod_extensions/ext_physics/rigid-bodies.html#collision-settings