isaac-sim/IsaacLab #70: [Proposal] Enabling Omniverse Streaming Client

Mayankm96, 2023-04-17:
### Proposal

Omniverse Streaming Client is a lightweight application allowing Users to connect to Omniverse applications deployed on the network or over the Internet. 

> https://docs.omniverse.nvidia.com/app_streaming-client/app_streaming-client/overview.html

### Motivation

It is quite useful for people who are developing on remote servers. Thus, it would help to have a pipeline to enable this functionality easily.

### Pitch

* To enable live stream, the user needs to load the default app experience in their standalone python scripts
* After that, adding the following piece of code to the script should enable live streaming:

```python
import carb
from omni.isaac.core.utils.extensions import enable_extension
from omni.isaac.core.utils.carb import set_carb_setting

# settings interface
carb_settings = carb.settings.get_settings()
# set values
set_carb_setting(carb_settings, "/app/livestream/enabled", True)
set_carb_setting(carb_settings, "/app/window/drawMouse", True)
set_carb_setting(carb_settings, "/app/livestream/proto", "ws")
set_carb_setting(carb_settings, "/app/livestream/websocket/framerate_limit", 120)
set_carb_setting(carb_settings, "/ngx/enabled", False)
# enable extensions for live stream
enable_extension("omni.kit.livestream.native")
enable_extension("omni.services.streaming.manager")
```

### Additional context

[OmniIsaacGymEnvs](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs) has a way to enable live stream easily form the command line.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

neale, 2023-04-18:
This worked out of the box for me in the demo environments :)
In the Sb3 / rlgames training scripts I found that the livestreaming was extremely slow -> glacial depending on the task. 
Could be a networking issue, ill check again tomorrow.

Mayankm96, 2023-04-18:
Yes, training with live streaming enabled is the same as running without headless. This adds a bunch of overhead because multiple kit extensions are being stepped at every app update call.

From what I tested using the `Isaac-Franka-Reach-v0` environment:
* Running headless without viewport enabled: ~65,000 FPS
* Running headless with viewport enabled: ~57,000 FPS
* Running with GUI: ~13,000 FPS

AutonomousHansen, 2023-04-24:
Hello,

I am currently attempting to get livestreaming working from some demo scripts after making the above addition, but am encountering segfaults whenever I actually attempt to connect the streaming client.

For context, I am running out of the IsaacSim Docker container on a Google Cloud VM. When I run `runheadless.native.sh` I am able to stream as expected. It seems the steps I'm running should be identical to yours, and I can't seem to find anything in the readout which specifies the problem, so I'm not sure why I get this mysterious segfault. I would appreciate your insight! Thank you.

Here is the relevant excerpt of play_arms.py (the rest is unchanged):

```
...
# launch omniverse app
config = {"headless": args_cli.headless}
simulation_app = SimulationApp(config)

import carb
from omni.isaac.core.utils.extensions import enable_extension
from omni.isaac.core.utils.carb import set_carb_setting

# settings interface
carb_settings = carb.settings.get_settings()
# set values
set_carb_setting(carb_settings, "/app/livestream/enabled", True)
set_carb_setting(carb_settings, "/app/window/drawMouse", True)
set_carb_setting(carb_settings, "/app/livestream/proto", "ws")
set_carb_setting(carb_settings, "/app/livestream/websocket/framerate_limit", 120)
set_carb_setting(carb_settings, "/ngx/enabled", False)
# enable extensions for live stream
enable_extension("omni.kit.livestream.native")
enable_extension("omni.services.streaming.manager")

"""Rest everything follows."""

import torch

import omni.isaac.core.utils.prims as prim_utils
from omni.isaac.core.simulation_context import SimulationContext
from omni.isaac.core.utils.viewports import set_camera_view

import omni.isaac.orbit.utils.kit as kit_utils
from omni.isaac.orbit.robots.config.franka import FRANKA_PANDA_ARM_WITH_PANDA_HAND_CFG
from omni.isaac.orbit.robots.config.universal_robots import UR10_CFG
from omni.isaac.orbit.robots.single_arm import SingleArmManipulator
from omni.isaac.orbit.utils.assets import ISAAC_NUCLEUS_DIR

"""
Main
"""

def main():
...
```
And here is the command, and the resulting terminal readout and error:
```
root@isaac-sim-1:/isaac-sim/orbit/source/standalone/demo# orbit -p play_arms.py --headless
[INFO] Using python from: /isaac-sim/python.sh                                                                                             
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.isaac.kit.app_framework'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/isaac-sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/isaac-sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/isaac-sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/isaac-sim/exts', '--ext-folder', '/isaac-sim/apps', '--/physics/cudaDevice=0', '--portable', '--no-window', '--allow-root']
Passing the following args to the base kit application:  ['--headless']
[Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/2022.2/kit_20230424_165813.log
2023-04-24 16:58:13 [18ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.lidar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.sensors.nv.lidar' or '/isaac-sim/exts/omni.sensors.nv.lidar/config'
2023-04-24 16:58:13 [18ms] [Warning] [omni.ext.plugin] [ext: omni.sensors.nv.radar] Extensions config 'extension.toml' doesn't exist '/isaac-sim/exts/omni.sensors.nv.radar' or '/isaac-sim/exts/omni.sensors.nv.radar/config'
[0.181s] [ext: omni.stats-0.0.0] startup
[0.219s] [ext: omni.rtx.shadercache-1.0.0] startup
[0.232s] [ext: omni.assets.plugins-0.0.0] startup
[0.235s] [ext: omni.gpu_foundation-0.0.0] startup
[0.246s] [ext: carb.windowing.plugins-1.0.0] startup
2023-04-24 16:58:13 [234ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2023-04-24 16:58:13 [234ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.3]) (impl: carb.windowing-glfw.plugin)
[... 292 lines cut ...]
[15.562s] [ext: omni.kit.livestream.native-0.1.5] startup

Active user not found. Using default user [kiosk]Streaming server started.
[16.085s] [ext: omni.services.facilities.base-1.0.2] startup
[16.100s] [ext: omni.services.core-1.3.0] startup
[16.262s] [ext: omni.services.transport.server.base-1.1.0] startup
[16.264s] [ext: omni.services.transport.server.http-1.1.2] startup
[16.283s] [ext: omni.services.streaming.manager-0.3.6] startup
2023-04-24 16:58:29 [16,360ms] [Warning] [omni.isaac.core.utils.nucleus] /persistent/app/omniverse/mountedDrives setting not found
2023-04-24 16:58:29 [16,610ms] [Warning] [omni.isaac.core.utils.nucleus] /persistent/app/omniverse/mountedDrives setting not found
2023-04-24 16:58:29 [16,611ms] [Warning] [omni.isaac.core.utils.nucleus] /persistent/app/omniverse/mountedDrives setting not found
[... cut ...]
[INFO]: Setup complete...
[INFO]: Resetting robots state...
[INFO]: [Step 000]: Flipping gripper command...
[INFO]: [Step 200]: Flipping gripper command...
[INFO]: [Step 400]: Flipping gripper command...
[INFO]: [Step 600]: Flipping gripper command...
[INFO]: [Step 800]: Flipping gripper command...
[INFO]: Resetting robots state...
[INFO]: [Step 000]: Flipping gripper command...
/isaac-sim/python.sh: line 41:  1089 Segmentation fault      (core dumped) $python_exe "$@" $args
There was an error running python
```

AutonomousHansen, 2023-04-25:
I was able to get this to run using the script above after applying this fix to omni.isaac.sim.python.kit in the /apps folder: 
In line # 550, change to:
“omni.isaac.quadruped” = {order = 10}

Source: https://forums.developer.nvidia.com/t/running-a-standalone-example-with-gui-in-docker-container/248147/6?u=hhansen

Mayankm96, 2023-04-25:
Thanks for finding this fix! It seems that it is better for Orbit to have its own set of "app" configuration files. Just to help us patch these kinds of issues easily.

AutonomousHansen, 2023-04-26:
> Thanks for finding this fix! It seems that it is better for Orbit to have its own set of "app" configuration files. Just to help us patch these kinds of issues easily.

Sure thing! And I totally agree, no reason to make patching more difficult than it needs to be.

AutonomousHansen, 2023-04-27:
For anyone attempting this with ROS--
The client will crash on a connection attempt if you load the ros_bridge extension before omni.kit.livestream.native. If you load it afterwards, it will behave as expected.

Mayankm96, 2023-12-19:
Closing this issue as we have added livestream options on the devel branch. For more details, please check the documentation.

closed 2023-12-19 by Mayankm96

THSID2023, 2024-03-16:
Does this work with workstation version of nucleus? or is it just for enterprise version?