isaac-sim/IsaacLab #71: [Proposal] gamepad support as teleoperation device

yangcyself, 2023-04-18:
### Proposal

I propose adding the gamepad as a device for teleoperation.

### Motivation

A gamepad will make it easier for the teleoperation of the robot arms than the keyboard. Supporting the gamepad also makes the feature set more comprehensive.

### Pitch

A clear and concise description of what you want to happen.

### Alternatives

No

### Additional context

Gamepad has already been used for camera navigation. So users need to unselect the "Gamepad Camera Control" option in the GUI before using it for teleoperation.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

Mayankm96, 2023-04-18:
Good point. Maybe we can use the Gampepad Manipulation from Omniverse?

Some useful pointers for this:

* https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/ext_omni_isaac_gamepad.html
* https://docs.omniverse.nvidia.com/prod_kit/prod_kit/programmer_ref/input-devices/gamepad.html

yangcyself, 2023-04-18:
Yes, I already tried. I'm creating a draft PR.

Mayankm96, 2023-04-18:
Awesome! Looking forward to it :)

PR #72 (2023-04-23): Adds Gamepad device support for teleoperation
# Description

Add implementation of `Se3Gamepad` and `Se2Gamepad`.
Add an option of `gamepad` into `teleop_se3_agent.py`

Fixes #71 

## Type of change

- New feature (non-breaking change which adds functionality)

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with `./orbit.sh --format`
- [x] I have made corresponding changes to the documentation
- [x] My changes generate no new warnings
- [ ] I have added tests that prove my fix is effective or that my feature works
- [x] I have updated the changelog and the co [... cut]
closed 2023-04-23 by Mayankm96