isaac-sim/IsaacLab #75: tethering object to the gripper in LeggedMobileManipulator

dengyh16code, 2023-04-20:
We’re trying to implement a simple gripper that works not by grasp friction but by tethering the object to the gripper when the gripper is close enough. I have used class LeggedMobileManipulator in issac orbit. I found there is a class named SurfaceGripper (https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.surface_gripper/docs/index.html) can solve this problem. But how can I add SurfaceGripper to LeggedMobileManipulator in Issac orbit. And are there other solutions that can achieve this goal?

Mayankm96, 2023-04-20:
You can probably make it as an actuator group and it through that.

Mayankm96, 2023-05-26:
Closing the issue due to no activity. Please re-open if still unresolved.

closed 2023-05-26 by Mayankm96