isaac-sim/IsaacLab #76: [Proposal] Compositional Configuration via Dataclasses (Hydra, etc.)

romesco, 2023-04-24:
### Proposal

Orbit has already made thoughtful design decisions that support really nice UX/UI paradigms for configuration management. Showing a few use-cases of how this might be done would be beneficial for the community as it can be difficult to decide "how to start". We would like to have one or two best-practices examples that fully leverage configuration via `dataclasses`. Preferably, one which makes no commitment to configuration framework, and one which demonstrates a a few tweaks that can allow Hydra to be used at its most powerful capacity.

### Pitch

We'll provide an example that shows how to set up an environment(robot / scene), RL algorithm, and training script and configure all of this via dataclasses.

We'll then show how to extend this configuration paradigm via some key features of Hydra.

### Alternatives

We can also look into what benefits other configuration frameworks have (one such being tyro), as long as they use `dataclasses` as the principle data structure.

### Additional context

#57 and #67 is a related issue/PR pair, and it is a good start! We will build on this and provide more use-cases that leverage Hydra and are clean / easy to understand.

### Checklist

- [x ] I have checked that there is no similar issue in the repo (**required**)

romesco, 2023-04-24:
Following this proposal, I'll update with a working draft in PRs which we can look over and provide comments on.

Mayankm96, 2023-04-24:
Thanks a lot for this! @romesco 

Looking forward to your proposal :)

Mayankm96, 2023-04-25:
Adding reference to this issue [IGE #127](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/pull/127) on auto-completion when using Hydra. Seems pretty handy.

Mayankm96, 2023-05-26:
Hi @romesco, I am wondering if there any updates on the plan with Hydra integration :)

romesco, 2023-07-05:
Thanks for the tag! I have something I'll push this weekend! Sorry about the tardiness, I had conference travel and then a deadline!

Mayankm96, 2024-09-06:
Hi, since the number of issues has been rising, we decided to go over the backlogs and remove the stale ones.

We did an initial Hydra integration, as you already saw in: https://github.com/isaac-sim/IsaacLab/pull/700 

You had some great suggestions there for which it would be interesting to adapt the code. If you have the availability, please feel free to submit a new issue so that we can start working more actively. Thank you for the support! :)

closed 2024-09-06 by Mayankm96