isaac-sim/IsaacLab #80: [Question] How to create a visual mesh?

PARKBONG, 2023-05-03:
### Question

Hi! I want to create a visual-mesh like [this](https://isaac-orbit.github.io/videos/in_hand/allegro_hand_orbit.m4v).

Unfortunately, `In-hand Manipulation` env is not opened yet, I want to edit the existing code. With these option,
- collision_enabled=False
- disable_gravity=True
I expected the same behavior in the [picture](https://isaac-orbit.github.io/videos/in_hand/allegro_hand_orbit.m4v).
**However, the `visual_objects` still have a collider, while gravity is disabled.**
Can you give me some advice?

Thanks!
**Reproduce Steps======================================================**

I just added some lines to the `.cfg` file.
**[lift_cfg.py](https://github.com/NVIDIA-Omniverse/Orbit/blob/8e42f0574792f971a44459a061dd3704cac38bb1/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/manipulation/lift/lift_cfg.py#L29)**
```
@configclass
class VisualObjectCfg(RigidObjectCfg):

    meta_info = RigidObjectCfg.MetaInfoCfg(
        usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Blocks/DexCube/dex_cube_instanceable.usd",  scale=(1, 1, 1))
    init_state = RigidObjectCfg.InitialStateCfg(
        pos=(0.4, 0.0, 0.075), rot=(1.0, 0.0, 0.0, 0.0), lin_vel=(0.0, 0.0, 0.0), ang_vel=(0.0, 0.0, 0.0))
    collision_props = RigidObjectCfg.CollisionPropertiesCfg(collision_enabled=False)
    rigid_props = RigidObjectCfg.RigidBodyPropertiesCfg(disable_gravity=True)

...

@configclass
class LiftEnvCfg(IsaacEnvCfg):
   ...
    # -- object
    object: ManipulationObjectCfg = ManipulationObjectCfg()
    # -- VISUAL OBJECT 
    visual_object: VisualObjectCfg = VisualObjectCfg()
    ...
```

After that, I added two lines one my `env` file.
**[lift_env.py](https://github.com/NVIDIA-Omniverse/Orbit/blob/8e42f0574792f971a44459a061dd3704cac38bb1/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/manipulation/lift/lift_env.py#L38)**

```
class LiftEnv(IsaacEnv):

    def __init__(self, cfg: LiftEnvCfg = None, headless: bool = False):
        ...
        self.object = RigidObject(cfg=self.cfg.object)
        self.visual_object = RigidObject(cfg=self.cfg.visual_object) # Here
        ...

    def _design_scene(self) -> List[str]:
        ...
        self.object.spawn(self.template_env_ns + "/Object")
        self.visual_object.spawn(self.template_env_ns + "/VisualObject") # Here
        ...
```

Mayankm96, 2023-05-26:
That is a good question. I have an idea on how to do this but it would take some time for me to get to it.

Basically, you can create a new rigid body and set its collision flag to be (N_envs + 1). This way it would be maksed out from collisions with all other bodies in the scene. That is how we have been doing in the past (in IsaacGym and the video you tagged).

Mayankm96, 2024-04-21:
We are glad to inform that in-hand manipulation environment was added in v0.3. You can find it here: https://github.com/NVIDIA-Omniverse/orbit/tree/main/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/manipulation/inhand

We use the VisualizationMarkers here for the goal markers you asked above: https://isaac-orbit.github.io/orbit/source/api/orbit/omni.isaac.orbit.markers.html#omni.isaac.orbit.markers.VisualizationMarkers

closed 2024-04-21 by Mayankm96