isaac-sim/IsaacLab #82: [Question] Where to find other environments from paper and doc?

Passworteingeben, 2023-05-08:
### Question
Hello Orbit Team,

I went through the docs and were able to run the environments that I was able to find namely Cartpole, Ant, Humanoid, Velocity, Reach and Lift and I was wondering where the other examples showcased in your paper and docs (here: https://isaac-orbit.github.io/orbit/source/features/environments.html) are and how to access them. 

The base examples only show the basic observations like the position and orientation of the actor, object, and target position, however a more complex example that utilizes more complex observations would be appreciated.

For example Cloth Folding or Mobile Reach would be interesting, as these require observations which would require the NN to learn spacial awareness.

I would greatly appreciate it if you publish some of the more complex examples as Isaac still has a quite meager community support for how exactly to utilize the engine to its potential.

Kind regards

BruceGeLi, 2023-09-04:
Hi, are there any updates for the other environments?
I read your paper, install the code base, get the demo work, and then I realize that I could not find those fancy envs...which is a little bit disappointing...
 
:(

duidui6666, 2024-04-09:
Could you please add it now? I want to try running the mobile manipulator task of opening the cabinet in the paper.

closed 2024-07-29 by Mayankm96