isaac-sim/IsaacLab #87: [Bug Report] Oscillatory motion of ridgeback panda with cuda device

harshmahesheka, 2023-05-23:
### Bug Description
If you try to run play_rigdeback_franka.py with the device as cuda in the simulation context it produces weird behavior in which the bot turns clockwise and starts oscillating around the perpendicular axis as soon as in the video below. 
### Steps to reproduce
In the [line](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/standalone/demo/play_ridgeback_franka.py#LL81C71-L81C71) change   ` sim = SimulationContext(physics_dt=0.01, rendering_dt=0.01, backend="torch")`       to     ` sim = SimulationContext(physics_dt=0.01, rendering_dt=0.01, backend="torch", device="cuda:0")` and run the file using the command  ` ./orbit.sh -p source/standalone/demo/play_ridgeback_franka.py `
### System Info

- Commit: ce8c1c015f187790ed6e894cf66bbb26abc13cb7
- Isaac Sim Version: 2022.2.1
- OS: 20.04
- GPU:  RTX A6000 
- CUDA: 11.7 
- GPU Driver: 515.105.01

### Additional context
Video of error- [Link](https://drive.google.com/file/d/1SOeEt4Q5cGHTf3p1XqZxqaSY43NewPd5/view?usp=sharing)

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2023-05-25:
Hi @harshmahesheka,

We identified a regression in GPU PhysX simulation with the Isaac Sim 2022.2.1 release that is causing issues in ArticulationView classes. I have a feeling that the same issue might be the cause of the problem here.

If it isn't too inconvenient, can you check with Isaac Sim 2022.2.0 release to see if it behaves weirdly there or not.

51616, 2023-06-01:
I have the same problem with Isaac Sim 2022.2.0.

51616, 2023-06-12:
I tried lowering the mass of the base of the robot and it seemed like it fixed the problem when simulating a few robots. The problem still persists for some env instances if the robot is cloned for many instances. Specifically, when using only 1 or 2 instances, the problem doesn't exist but it happens when I clone 128 instances of the robot.

51616, 2023-06-13:
I found a workaround by disabling self-collision of the robot. You can change the config file `source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/config/ridgeback_franka.py` by adding

```
...
from ..robot_base_cfg import RobotBaseCfg
...

RIDGEBACK_FRANKA_PANDA_CFG = MobileManipulatorCfg(
    ...
    articulation_props=RobotBaseCfg.ArticulationRootPropertiesCfg(
        enable_self_collisions=False
    ),
   ...
```

Hope this helps :)

closed 2024-02-10 by Mayankm96