isaac-sim/IsaacLab #94: [Bug Report] Orbit does not work with Isaac Sim 2022.2.1

julesser, 2023-06-28:
### Describe the bug
Hi there! Seems as if Orbit currently does not work with the latest version of Isaac SIM (2022.2.1). A similiar bug as described below already has been reported in [this forum post](https://forums.developer.nvidia.com/t/orbit-does-not-work-with-isaac-sim-2022-2-1/251572/1), so I'm catching up here to resolve it. 
 
### Steps to reproduce
Starting off with a fresh install of Orbit following the documentation and Isaac SIM 2022.2.1,  `./orbit.sh --install` crashes with the following error (see [log.txt](https://github.com/NVIDIA-Omniverse/Orbit/files/11890785/log.txt) for the full console output):

```
The above exception was the direct cause of the following exception:
  
  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-0f357ehf/gym_e0af3fd524a74877b879116de6a6d174/setup.py", line 74, in <module>
      "Programming Language :: Python :: 3.9",
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.pip_archive/pip_prebundle/setuptools/__init__.py", line 87, in setup
      return distutils.core.setup(**attrs)
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.pip_archive/pip_prebundle/setuptools/_distutils/core.py", line 185, in setup
      return run_commands(dist)
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.pip_archive/pip_prebundle/setuptools/_distutils/core.py", line 201, in run_commands
      dist.run_commands()
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.pip_archive/pip_prebundle/setuptools/_distutils/dist.py", line 969, in run_commands
      self.run_command(cmd)
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.pip_archive/pip_prebundle/setuptools/dist.py", line 1217, in run_command
      super().run_command(command)
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extscore/omni.kit.pip_archive/pip_prebundle/setuptools/_distutils/dist.py", line 988, in run_command
      cmd_obj.run()
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 395, in run
      self.egg2dist(self.egginfo_dir, distinfo_dir)
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/wheel/bdist_wheel.py", line 534, in egg2dist
      pkg_info = pkginfo_to_metadata(egginfo_path, pkginfo_path)
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/wheel/metadata.py", line 160, in pkginfo_to_metadata
      for key, value in generate_requirements({extra: reqs}):
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/wheel/metadata.py", line 138, in generate_requirements
      for new_req in convert_requirements(depends):
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/wheel/metadata.py", line 103, in convert_requirements
      parsed_requirement = Requirement(req)
    File "/home/admin-jesser/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/wheel/vendored/packaging/requirements.py", line 37, in __init__
      raise InvalidRequirement(str(e)) from e
  wheel.vendored.packaging.requirements.InvalidRequirement: Expected end or semicolon (after version specifier)
      opencv-python>=3.
                   ~~~^
  ----------------------------------------
  ERROR: Failed building wheel for gym

```

To confirm that it's not related to my system: When using Isaac SIM 2022.2.0 instead (`ISAACSIM_PATH="${HOME}/.local/share/ov/pkg/isaac_sim-2022.2.0"`),  the installation of extensions `./orbit.sh --install` runs without any errors. 

### System Info

Describe the characteristic of your environment:

- Commit: [f2d97bd](https://github.com/NVIDIA-Omniverse/Orbit/commit/f2d97bdcddb3005d17d0ebd1546c7064bc7ae8bc) (latest)
- Isaac Sim Version: 2022.2.1
- OS: Ubuntu 22.04
- GPU: A5000
- CUDA: 12.0
- GPU Driver: 525.60.11

### Checklist

- [x ] I have checked that there is no similar issue in the repo (**required**)
- [x ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2023-06-28:
Hi @julesser

The issue is coming due to latest upgrade in setuptools from Python that is causing the underlying OpenAI Gym dependency to fail.

Can you try the solution mentioned here as a fix: https://isaac-orbit.github.io/orbit/source/refs/issues.html#installation-errors-due-to-gym-0-21-0

julesser, 2023-06-28:
Hi  @Mayankm96  Thanks for the hint! 

After adding `./orbit.sh -p -m pip install -U setuptools==66` the installation also runs through for 2022.2.1.

closed 2023-06-28 by julesser

942411526, 2023-11-23:
my version is 2023.1.0 also have this problem the setuptools=66.