isaac-sim/IsaacLab #96: [Bug Report] Kit Util Error when Importing USD with Physx Collider Meshes

jessicah7654, 2023-07-12:
### Describe the bug

Error is thrown when trying to use '''set_collision_properties()'''' in omni.isaac.orbit>utils>kit.
''''set_collision_properties()''' is trying to use ''''PhysxSchema.PhysxCollisionAPI''''s '''GetCollisionEnabledAttr()'''' method, which doesn't exist.

### Steps to reproduce

Open the existing panda_instanceable.usd from the Orbit nucleus in the Isaacsim GUI application.
Edit the USD file by deleting _collisions_ xform and mesh from each panda link in Stage tree.
For each _visuals_ xform for each panda link, Add>Physics>Colliders to create collider meshes.
Save a new USD file.

Use this new USD file in the play_arms pipeline by editing the USD file path in the franka.py file.
The file should throw an error because the new Physxcollision meshes now have to be handled.

```
Exception has occurred: AttributeError
'PhysxCollisionAPI' object has no attribute 'GetCollisionEnabledAttr'
  File "/home/jess/Research/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/kit.py", line 432, in set_collision_properties
    physx_collision_api.GetCollisionEnabledAttr().Set(collision_enabled)
  File "/home/jess/Research/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/kit.py", line 594, in set_nested_collision_properties
    set_collision_properties(prim_utils.get_prim_path(child_prim), **kwargs)
  File "/home/jess/Research/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/BRL_manipulator/BRL_manipulator.py", line 90, in spawn
    kit_utils.set_nested_collision_properties(prim_path, collision_enabled = True, contact_offset=0.02, rest_offset=0.0)
  File "/home/jess/Research/orbit/source/standalone/demo/play_arms_BRL.py", line 96, in main
    robot.spawn("/World/Robot_1", translation=(0.0, -1.0, 0.0))
  File "/home/jess/Research/orbit/source/standalone/demo/play_arms_BRL.py", line 190, in <module>
    main()
AttributeError: 'PhysxCollisionAPI' object has no attribute 'GetCollisionEnabledAttr'
```
(play_arms_BRL.py and BRL_manipulator.py are adaptations of play_arms.py and single_arm.py for our manipulator robot)

### System Info

Describe the characteristic of your environment:

- Commit: 1ee7a26 or 0c6a23b
- Isaac Sim Version: 2022.2.0
- OS: Ubuntu 22.04
- GPU: GeForce RTX™ 3060
- CUDA: 11.7
- GPU Driver: 515.105.01

### Additional context

Problem solved if you remove the lines 
'''
if collision_enabled is not None:
     physx_collision_api.GetCollisionEnabledAttr().Set(collision_enabled)
'''

### Checklist

- [X] I have checked that there is no similar issue in the repo (**required**)
- [X] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2023-07-15:
Hi, can you share the USD file itself?

cold-young, 2024-01-23:
I found this bug too. 
I think fix "physx_collision_api = UsdPhysics.CollisionAPI"
you can find "enable_collisioin" function of "geometry_prim_view.py" in isaacsim 2023.1.0.hotfix.1

Mayankm96, 2024-04-26:
In the release after 0.1, we handle this behavior properly now. Please check here for further details: https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/schemas/schemas.py#L353-L361

Closing this issue now since it is solved. Thank you!

closed 2024-04-26 by Mayankm96