isaac-sim/IsaacLab #97: Checking Collision between Object and Franka Robot

LenaP7300, 2023-07-18:
Hello,

I am searching for a function to check for collision between the Franka Robot and a Rigid Object. More concretely, a function that tells the Environment when a collision has occurred. I could not find any such function under omni.isaac.robot or omni.isaac.objects.rigid. 
Am I looking in the wrong place or missing something, or does something like this not exist?

Best regards.

Mayankm96, 2023-07-18:
You can use the RigidContactView from Isaac Sim for this: https://docs.omniverse.nvidia.com/isaacsim/latest/ext_omni_isaac_core.html#create-rigidcontactview

LenaP7300, 2023-07-20:
Thank you very much for the tip!
I tested a few things based on your example. The RigidObject has a RigidPrimView cylinder.objects. I called cylinder.objects.get_net_contact_forces() after setting track_contact_forces=True. (I had to adjust the implementation of cylinder.objects for this to work.) I now get a valid matrix, but whenever the robot hits the cylinder this error appears:

2023-07-20 19:47:29 [19,072ms] [Warning] [omni.hydra.scene_delegate.plugin] InstanceAdapter - cannot find cache item for proto /World/envs/env_0/Cylinder/collisions.proto_collisions_id0
2023-07-20 19:47:29 [19,072ms] [Warning] [omni.hydra.scene_delegate.plugin] InstanceAdapter - cannot find cache item for proto /World/envs/env_0/Cylinder/visuals.proto_visuals_id0

The Franka Robot then glitches into the table.

If I use more than one environment self.cylinder.initialize(self.env_ns + "/.*/Cylinder", track_contact_forces=True, prepare_contact_sensors=True) raises the exception "Failed to set rigid body transforms in backend"

Both faults only appear when track_contact_forces is true, while everything else remains unchanged.
If you have any ideas on how I could solve this, I would appreciate them. Otherwise, thank you for your time!

Mayankm96, 2023-08-02:
Is it possible to get a repro for this? Maybe it is something to do with the tuning of collision offsets.

LenaP7300, 2023-08-02:
The errors I am getting are the same as in [this post ](https://forums.developer.nvidia.com/t/errors-with-hydra-scene-graph-instancing/257640). You can find the code for the environment in which the errors appear [here](https://github.com/LenaP7300/Obstacle/tree/main/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/manipulation/obstacle).

2361098148, 2023-08-12:
I am facing the same problem, Have you figure out how to solve the problem?

LenaP7300, 2023-08-13:
I figured it out. I applied the ContactReporter in the function initialize_views, but I think that it creates a prim or something. Orbit seems to have some trouble adding them anywhere other than design_scene. As soon as I added it in design_scene, it worked.

2361098148, 2023-08-14:
Could you share the gist, I would appreciate. Do you mean add them here:
```
    def _design_scene(self) -> List[str]:
        # ground plane
        if self.cfg.terrain.use_default_ground_plane:
            # use the default ground plane
            kit_utils.create_ground_plane(
                "/World/defaultGroundPlane",
                static_friction=1.0,
                dynamic_friction=1.0,
                restitution=0.0,
                improve_patch_friction=True,
                combine_mode="max",
            )
        else:
            prim_utils.create_prim("/World/defaultGroundPlane", usd_path=self.cfg.terrain.usd_path)

        # robot
        self.robot.spawn(self.template_env_ns + "/Robot")

        # setup debug visualization
        if self.cfg.viewer.debug_vis and self.enable_render:
            # create point instancer for foot contact
            self._feet_contact_marker = PointMarker(
                "/Visuals/feet_contact", self.num_envs * len(self.cfg.robot.feet_info), radius=0.035
            )
            # create point instancer to visualize the goal base velocity
            self._base_vel_goal_markers = StaticMarker(
                "/Visuals/base_vel_goal",
                self.num_envs,
                usd_path=self.cfg.marker.usd_path,
                scale=self.cfg.marker.scale,
                color=(1.0, 0.0, 0.0),
            )
            # create marker for viewing current base velocity
            self._base_vel_markers = StaticMarker(
                "/Visuals/base_vel",
                self.num_envs,
                usd_path=self.cfg.marker.usd_path,
                scale=self.cfg.marker.scale,
                color=(0.0, 0.0, 1.0),
            )
        # return list of global prims
        return ["/World/defaultGroundPlane"]
```
Could you tell me how, I'm not quite sure yet

2361098148, 2023-08-18:
> RigidContactView
Please, could you give me a hand. I have tried but failed, could you give me more details

LenaP7300, 2023-08-21:
In __init__() 

# subscribe to physics contact report event, this callback issued after each simulation step
        self._contact_report_sub = get_physx_simulation_interface().subscribe_contact_report_events(self._on_contact_report_event)

def _design_scene(self):
        # ground plane
        kit_utils.create_ground_plane("/World/defaultGroundPlane", z_position=-1.05)
        # table
        prim_utils.create_prim(self.template_env_ns + "/Table", usd_path=self.cfg.table.usd_path)
        # cube
        self.cube.spawn(self.template_env_ns + "/Cube")
        cubePrim = prim_utils.get_prim_at_path(self.template_env_ns + "/Cube")
        PhysxSchema.PhysxContactReportAPI.Apply(cubePrim)
        # robot
        self.robot.spawn(self.template_env_ns + "/Robot")
        # camera
        self.camera.spawn(self.template_env_ns + "/Camera", translation=[1.5, 1.5, 1.5], orientation=[0.0, 0.0, 0.0, 0.0])
        set_camera_view([3.0, 1.0, 1.3], [0.0, 0.0, 0.0], self.camera.prim_path)

        # setup debug visualization
        if self.cfg.viewer.debug_vis and self.enable_render:
            # create point instancer to visualize the goal points
            self._goal_markers = PointMarker("/Visuals/ee_goal", self.num_envs, radius=0.05)
            # create marker for viewing end-effector pose
            self._ee_markers = StaticMarker(
                "/Visuals/ee_current", self.num_envs, usd_path=self.cfg.marker.usd_path, scale=self.cfg.marker.scale
            )
            # create marker for viewing command (if task-space controller is used)
            if self.cfg.control.control_type == "inverse_kinematics":
                self._cmd_markers = StaticMarker(
                    "/Visuals/ik_command", self.num_envs, usd_path=self.cfg.marker.usd_path, scale=self.cfg.marker.scale
                )
        # return list of global prims
        return ["/World/defaultGroundPlane"]

I did not end up using RigidPrimView, but the ContactReporter as in the example [here](https://docs.omniverse.nvidia.com/kit/docs/omni_physics/latest/source/extensions/omni.physx/docs/index.html). I hope this helps.

2361098148, 2023-08-29:
Thanks for your reply! My problem is solved by the method proposed in [issue110](https://github.com/NVIDIA-Omniverse/Orbit/issues/110#issuecomment-1696834920), finally.

notFoundThisPerson, 2024-06-26:
Hello, is there any API that reports which object is in collision with the robot? The RigidContactView seems only returns the contact force vector of the links without the object information, e.g. I want to check whether the specified cube is in collsion with the robot, rather than something else.