isaac-sim/IsaacLab #101: how to compute inverse kinematics (IK) on robot arms in orbit? [not for control]

chauvinSimon, 2023-07-26:
## Question

### context

my **franka robot arms** derive from `SingleArmManipulator`, inspired from the **orbit** example: 
https://isaac-orbit.github.io/orbit/source/tutorials/01_arms.html?highlight=singlearmmanipulator

---

### what I would like

I need a `Inverse Kinematics` function with:
- inputs = desired EE pose (cartesian position and orientation)
- outputs = joint positions (7 angles in the case of the franka arms) corresponding to the desired EE pose
  - _ok, there may be multiple solutions_

:warning: **I do not need a IK controller!!** Rather a standalone function that can be call at any time, in any position and that directly returns the result, without moving the robots.

---

### what I have tried

- in `orbit`: I have tried with the `DifferentialInverseKinematics` class, but it is meant for **control** (to reach a EE pose in **multiple steps**)
[Orbit/source/standalone/demo/play_ik_control.py](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/standalone/demo/play_ik_control.py)

- in `isaac-sim` (not `orbit`): I have found this Lula module, which does the job:
https://docs.omniverse.nvidia.com/isaacsim/latest/advanced_tutorials/tutorial_motion_generation_lula_kinematics.html

---

### my problem
I have tried to use the **Lula tools with the orbit robots**

- `ArticulationKinematicsSolver` (in the `isaac-sim` Lula module) expects a `robot_articulation` of type `Articulation`
  - e.g. the `Franka(Robot)` can be passed since it derived from `Robot(Articulation)`: `Orbit/_isaac_sim/exts/omni.isaac.franka/omni/isaac/franka/franka.py`
  - note: `ArticulationKinematicsSolver` is defined in `Orbit/_isaac_sim/exts/omni.isaac.motion_generation/omni/isaac/motion_generation/articulation_kinematics_solver.py`
- `SingleArmManipulator(RobotBase)` (in `orbit`) is not of type `Articulation` but has an `ArticulationView` attribute named `articulations`

- `ArticulationView` is not a `Articulation`!
  - Although they are defined in the same `isaac_sim` folder: `Orbit/_isaac_sim/exts/omni.isaac.core/omni/isaac/core/articulations`

- so instantiating `articulation_kinematics_solver = ArticulationKinematicsSolver(orbit_robot)` or `articulation_kinematics_solver = ArticulationKinematicsSolver(orbit_robot.articulations)` is not an option
  - it will complain for instance about `AttributeError: 'SingleArmManipulator' object has no attribute 'handles_initialized'`

---

### questions

- is there a way to use the `lula_kinematics` modules with orbit robots?
- or is there any way to create a simple IK function for orbit robots (in particular for franka)?

Mayankm96, 2023-07-26:
Hi @chauvinSimon 

Yes, currently LULA directly takes in the `Articulation` class and gets the data it needs from it. Since they aren't "concerned" with the parallelization of the controller, they don't use the view classes. Due to this, you see the issues above.

We kind of hack it through in Orbit to have an example here that shows how to integrate the RMPFlow controller from LULA library in a way that it expects. The class outputs the joint actions which are then sent to our robot classes:

* RmpFlow Class: https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/controllers/rmp_flow.py
* Example script: https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/standalone/demo/play_rmpflow.py

chauvinSimon, 2023-07-26:
thank you @Mayankm96 for the fast answer

chauvinSimon, 2023-07-26:
Sorry one more thing @Mayankm96 :

- the `RmpFlow` example you kindly shared is not directly outputing the joint-position corresponding to the _input desired EE pose_.

- Instead, it is used for **control()**: its `compute` method derives a `next_action`, i.e. the intermediate target q position, used for the next frame. And this needs to be repeated in a loop until the robot reaches the _input desired EE pose_, set with `set_command(rmp_commands)`.

### my question
do you know how I could **directly get the `joint-positions` corresponding to `any desired EE pose`, without moving the robot?** In one function call. It sounds like a **basic IK utils**, that would only requires the robot geometry.

### motivation

I need that joint positions to set `robot_cfg.init_state.dof_pos` to initialize my robot at a desired EE pose: 

```
    class InitialStateCfg:
        """Initial state of the robot."""

        ...

        # dof state
        dof_pos: Dict[str, float] = MISSING

```

Mayankm96, 2023-07-26:
Hi,

The script I shared was mainly supposed to be an example on how you can integrate a controller from lula with Orbit.

You would need to modify it to get the output from the other lula controller you mentioned earlier.

For just IK checks to get valid initial poses, I wonder if you can use something that's parallelized instead. From a quick search this might do the job: https://github.com/UM-ARM-Lab/pytorch_kinematics

Then you can run batched updates over the Jacobians until convergence.

chauvinSimon, 2023-07-27:
Thank you @Mayankm96

closed 2023-07-27 by chauvinSimon

catachiii, 2024-03-06:
@Mayankm96 

Hi Mayank, I need help getting the correct Jacobian for quadrupeds from Orbit or Isaac Sim. 

I was running the example Isaac-Velocity-Flat-Anymal-C-v0. After the step function, I tried to access jacobian through `env.scene.articulations['robot'].root_physx_view.get_jacobians()` and by accessing the FL_FOOT vel jacobian wrt joint vel by `[0, 4, 0:3, -12:]`, I found almost all entries are zero, which is wrong... 

I'm not sure what happened... this can be very helpful for IK. 

Thanks!