isaac-sim/IsaacLab #102: [Question] How to get force/torque for task space force control?

okada-masashi, 2023-08-01:
I am trying to control `SingleArmManipulator` using the [task-space force controller](https://isaac-orbit.github.io/orbit/source/features/motion_generators.html#task-space-controllers).
To do so, I need to get the force/torque that is applied to the end-effector, but the [SingleArmManipulatorData](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/single_arm/single_arm_data.py) does not provide an API to get this information. Is there a way to get force/torque?

I found the following comment in [legged_robot.py](https://github.com/NVIDIA-Omniverse/Orbit/blob/f2d97bdcddb3005d17d0ebd1546c7064bc7ae8bc/source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/legged_robot/legged_robot.py#L149-L150).
```
# TODO: contact forces -- Waiting for contact sensors in IsaacSim.
#   For now, use heuristics for flat terrain to say feet are in contact.
```
Is it impossible to get force/torque due to IsaacSim limitations?

kumar-sanjeeev, 2023-08-01:
Hello @okada-masashi,

I did the following to get the applied torque values at different joints of the arm, including the end-effector. 

As [SingleArmManipulator](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/single_arm/single_arm.py) is derived from the [RobotBase](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/robot_base.py) class, which has an `articulations` attribute that is an instance of [ArticulationView](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#module-omni.isaac.core.articulations). You can call the `get_applied_joint_efforts` method on the ArticulationView instance.

For example:
```python
robot = SingleArmManipulator(cfg=robot_cfg)
articulation_view = robot.articulations
applied_joint_efforts = articulation_view.get_applied_joint_efforts() # can pass the indices as per requirement

```

**Note**: you might get zero values in return, if you have used the [Implicit Actuator Model](https://isaac-orbit.github.io/orbit/source/api/orbit.actuators.model.html) while using the task space controller :  [differential_inverse_kinematics](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/controllers/differential_inverse_kinematics.py)

Mayankm96, 2023-08-01:
Hi, if you want to obtain the joint torque values with implicit actuator model (i.e. the PhysX PD controller), there is a flag called `enable_dof_force_sensors` in the [ArticulationView](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#omni.isaac.core.articulations.Articulation) that should give you joint torque sensing. I haven't really experimented with it yet to confirm whether the values obtained from it are realistic enough or not.

If you need the force sensing on the link of the robot, you can use the contact forces using the [RigidContactView](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#omni.isaac.core.prims.RigidContactView) class. This quantity is what the prior work, [Factory](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/main/isaacgymenvs/tasks/factory/factory_control.py), used in Isaac Gym for task-space control. However, that only gives the normal contact force and not really the torque acting on the link.

We have made a feature request to the PhysX team to provide a 6-DoF wrench sensing on the links. Will have to check if that is going to be in the coming release of Isaac Sim or not.

kumar-sanjeeev, 2023-08-01:
Hello @Mayankm96 

Thanks for pointing out the way to obtain the joint torque values with the implicit actuator model. I tried enabling `enable_dof_force_sensors` in the [ArticulationView](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#omni.isaac.core.articulations.Articulation), but I am still receiving zero values when calling the `get_applied_joint_efforts` method on the [ArticulationView](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html#omni.isaac.core.articulations.Articulation) instance. I made the changes marked as **#### changes** in the [play_ik_control](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/standalone/demo/play_ik_control.py) example available in the demo section of the repository to try what you have suggested.

**Edited code script that I have tried:**
```python
"""Launch Isaac Sim Simulator first."""

import argparse

from omni.isaac.kit import SimulationApp

# add argparse arguments
parser = argparse.ArgumentParser("Welcome to Orbit: Omniverse Robotics Environments!")
parser.add_argument("--headless", action="store_true", default=False, help="Force display off at all times.")
parser.add_argument("--robot", type=str, default="franka_panda", help="Name of the robot.")
parser.add_argument("--num_envs", type=int, default=128, help="Number of environments to spawn.")
args_cli = parser.parse_args()

# launch omniverse app
config = {"headless": args_cli.headless}
simulation_app = SimulationApp(config)

"""Rest everything follows."""

import torch

import omni.isaac.core.utils.prims as prim_utils
from omni.isaac.cloner import GridCloner
from omni.isaac.core.simulation_context import SimulationContext
from omni.isaac.core.utils.carb import set_carb_setting
from omni.isaac.core.utils.viewports import set_camera_view

import omni.isaac.orbit.utils.kit as kit_utils
from omni.isaac.orbit.controllers.differential_inverse_kinematics import (
    DifferentialInverseKinematics,
    DifferentialInverseKinematicsCfg,
)
from omni.isaac.orbit.markers import StaticMarker
from omni.isaac.orbit.robots.config.franka import FRANKA_PANDA_ARM_WITH_PANDA_HAND_CFG
from omni.isaac.orbit.robots.config.universal_robots import UR10_CFG
from omni.isaac.orbit.robots.single_arm import SingleArmManipulator
from omni.isaac.orbit.utils.assets import ISAAC_NUCLEUS_DIR

"""
Main
"""

def main():
    """Spawns a single-arm manipulator and applies commands through inverse kinematics control."""

    # Load kit helper
    sim = SimulationContext(physics_dt=0.01, rendering_dt=0.01, backend="torch", device="cuda:0")
    # Set main camera
    set_camera_view([2.5, 2.5, 2.5], [0.0, 0.0, 0.0])
    # Enable GPU pipeline and flatcache
    if sim.get_physics_context().use_gpu_pipeline:
        sim.get_physics_context().enable_flatcache(True)
    # Enable hydra scene-graph instancing
    set_carb_setting(sim._settings, "/persistent/omnihydra/useSceneGraphInstancing", True)

    # Create interface to clone the scene
    cloner = GridCloner(spacing=2.0)
    cloner.define_base_env("/World/envs")
    # Everything under the namespace "/World/envs/env_0" will be cloned
    prim_utils.define_prim("/World/envs/env_0")

    # Spawn things into stage
    # Markers
    ee_marker = StaticMarker("/Visuals/ee_current", count=args_cli.num_envs, scale=(0.1, 0.1, 0.1))
[... 146 lines cut ...]

        #### changes
        print(f"Applied Joint Torques: {articulation_view.get_applied_joint_efforts()}")
        ####

        # update sim-time
        sim_time += sim_dt
        count += 1
        # note: to deal with timeline events such as stopping, we need to check if the simulation is playing
        if sim.is_playing():
            # update buffers
            robot.update_buffers(sim_dt)
            # update marker positions
            ee_marker.set_world_poses(robot.data.ee_state_w[:, 0:3], robot.data.ee_state_w[:, 3:7])
            goal_marker.set_world_poses(ik_commands[:, 0:3] + envs_positions, ik_commands[:, 3:7])

if __name__ == "__main__":
    # Run IK example with Manipulator
    main()
    # Close the simulator
    simulation_app.close()

```

am i doing something wrong ?

Mayankm96, 2023-08-02:
HI @kumar-sanjeeev ,

Looking at the code. It seems that the Isaac Sim wrapper `ArticulationView` is not exposing the correct underlying values.

Can you try this instead for getting the joint efforts:

```python
# directly calls the underlying physx view to get the quantities
articulation_view._physics_view.get_force_sensor_forces()
```

kumar-sanjeeev, 2023-08-02:
Hi @Mayankm96,

I tried directly calling the physx view to get the joint efforts, but I am getting the following errors while running the code.

**Error:**
```python
  File "sanju_ws/playing_with_implicit_actuator.py", line 230, in <module>
    main()
  File "sanju_ws/playing_with_implicit_actuator.py", line 138, in main
    articulation_view._physics_view.get_force_sensor_forces()
  File "/home/ksu1rng/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extsPhysics/omni.physics.tensors-104.2.4-5.1/omni/physics/tensors/impl/api.py", line 512, in get_force_sensor_forces
    raise Exception("Articulation view contains no force sensors")
Exception: Articulation view contains no force sensors
```
While looking for the root cause of this issue, I found the [Force Sensor documentation](https://docs.omniverse.nvidia.com/isaacsim/latest/isaac_sim_sensors_physics_articulation_force.html#), where a similar approach was suggested. However, it was also mentioned to do this step: **Add > Physics > Articulation Force Sensor** before calling `articulation_view._physics_view.get_force_sensor_forces()`. As far as I know, in the current Orbit codebase, the Franka robot articulation has not added the `Articulation Force Sensor` API to it. This might be the reason for the error. To confirm this, I conducted a small experiment using **Isaac Sim GUI and script editor**. During this experiment, I added the **franka_instanceable** from the asset and applied the Articulation Force Sensor to it. Then, I tried to print the output of articulation_view._physics_view.get_force_sensor_forces(), but I received the same error as mentioned above.

Mayankm96, 2023-08-02:
Can you pass the flag to enable the sensor?

> https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/robots/robot_base.py#L170

```
self.articulations = ArticulationView(self._prim_paths_expr, reset_xform_properties=False, enable_dof_force_sensors=True)
```

kumar-sanjeeev, 2023-08-02:
I tried and changed my local `robot_base.py` file as per your suggestion. Still getting the same error.
**Error:**
```python
Traceback (most recent call last):
  File "sanju_ws/playing_with_implicit_actuator.py", line 231, in <module>
    main()
  File "sanju_ws/playing_with_implicit_actuator.py", line 215, in main
    print(f"Applied Joint Torques: {articulation_view._physics_view.get_force_sensor_forces()}")
  File "/home/ksu1rng/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extsPhysics/omni.physics.tensors-104.2.4-5.1/omni/physics/tensors/impl/api.py", line 512, in get_force_sensor_forces
    raise Exception("Articulation view contains no force sensors")
```

**Following Script used to check functionality:**
```python
"""Launch Isaac Sim Simulator first."""

import argparse

from omni.isaac.kit import SimulationApp

# add argparse arguments
parser = argparse.ArgumentParser("Welcome to Orbit: Omniverse Robotics Environments!")
parser.add_argument("--headless", action="store_true", default=False, help="Force display off at all times.")
parser.add_argument("--robot", type=str, default="franka_panda", help="Name of the robot.")
parser.add_argument("--num_envs", type=int, default=128, help="Number of environments to spawn.")
args_cli = parser.parse_args()

# launch omniverse app
config = {"headless": args_cli.headless}
simulation_app = SimulationApp(config)

"""Rest everything follows."""

import torch

import omni.isaac.core.utils.prims as prim_utils
from omni.isaac.cloner import GridCloner
from omni.isaac.core.simulation_context import SimulationContext
from omni.isaac.core.utils.carb import set_carb_setting
from omni.isaac.core.utils.viewports import set_camera_view

import omni.isaac.orbit.utils.kit as kit_utils
from omni.isaac.orbit.controllers.differential_inverse_kinematics import (
    DifferentialInverseKinematics,
    DifferentialInverseKinematicsCfg,
)
from omni.isaac.orbit.markers import StaticMarker
from omni.isaac.orbit.robots.config.franka import FRANKA_PANDA_ARM_WITH_PANDA_HAND_CFG
from omni.isaac.orbit.robots.config.universal_robots import UR10_CFG
from omni.isaac.orbit.robots.single_arm import SingleArmManipulator
from omni.isaac.orbit.utils.assets import ISAAC_NUCLEUS_DIR

"""
Main
"""

def main():
    """Spawns a single-arm manipulator and applies commands through inverse kinematics control."""

    # Load kit helper
    sim = SimulationContext(physics_dt=0.01, rendering_dt=0.01, backend="torch", device="cuda:0")
    # Set main camera
    set_camera_view([2.5, 2.5, 2.5], [0.0, 0.0, 0.0])
    # Enable GPU pipeline and flatcache
    if sim.get_physics_context().use_gpu_pipeline:
        sim.get_physics_context().enable_flatcache(True)
    # Enable hydra scene-graph instancing
    set_carb_setting(sim._settings, "/persistent/omnihydra/useSceneGraphInstancing", True)

    # Create interface to clone the scene
[... 153 lines cut ...]
        
        #### changes
        print(f"Applied Joint Torques: {articulation_view.get_applied_joint_efforts()}") # giving the Torque tensors with zero values
        print(f"Applied Joint Torques: {articulation_view._physics_view.get_force_sensor_forces()}") # throwing the error
        ####
        
        
        # update sim-time
        sim_time += sim_dt
        count += 1
        # note: to deal with timeline events such as stopping, we need to check if the simulation is playing
        if sim.is_playing():
            # update buffers
            robot.update_buffers(sim_dt)
            # update marker positions
            ee_marker.set_world_poses(robot.data.ee_state_w[:, 0:3], robot.data.ee_state_w[:, 3:7])
            goal_marker.set_world_poses(ik_commands[:, 0:3] + envs_positions, ik_commands[:, 3:7])

if __name__ == "__main__":
    # Run IK example with Manipulator
    main()
    # Close the simulator
    simulation_app.close()

```

okada-masashi, 2023-08-08:
Thanks for letting know me about RigidContactView.
I tried to get contact force by adding the following snippet in [play_ik_demo.py](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/standalone/demo/play_ik_control.py), but the behavior has changed so that the table sinks as shown in the attached animation.
Do I need any additional codes?

```python
rigid_contact_view = RigidContactView('/World/envs/env_0/Robot/panda_hand', filter_paths_expr=['/World/envs/env_0/Table'], apply_rigid_body_api=False)
rigid_contact_view.initialize()
``` 
[image]

Mayankm96, 2023-08-11:
@okada-masashi I am checking internally if we can get contact forces between rigid bodies and static colliders on GPU pipeline.

> https://nvidia-omniverse.github.io/PhysX/physx/5.2.1/docs/RigidBodyCollision.html#pxpairfilteringmode

2361098148, 2023-08-21:
> Thanks for letting know me about RigidContactView. I tried to get contact force by adding the following snippet in [...]

Have you figure out? Im facing the same problems.

JohannesPankert, 2023-11-15:
Are there any updates on the original issue? I would like to get the applied (or measured) joint torques when using implicit actuators. 

> Can you pass the flag to enable the sensor?
>
>
> ```
> self.articulations = ArticulationView(self._prim_paths_expr, reset_xform_properties=False, enable_dof_force_sensors=True)
> ```

Adding the sensors both in the articulation base class and in my robot class did not help. When querying `articulation_view.get_applied_joint_efforts()` I still get the error:

```
  File ..., line 590, in _apply_actuator_model
    print(f"forces and torques:\n{self.root_physx_view.get_force_sensor_forces()}")
  File "/home/johannes/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/extsPhysics/omni.physics.tensors-104.2.4-5.1/omni/physics/tensors/impl/api.py", line 512, in get_force_sensor_forces
    raise Exception("Articulation view contains no force sensors")
Exception: Articulation view contains no force sensors

```

yeshgodseagility, 2024-06-11:
@Mayankm96 It seems several things have changed and I'm not sure what the best way to get the measured force/torque at a joint is. The original suggestion of using `get_force_sensor_forces()` probably doesn't work because the force sensor API was removed? (https://docs.omniverse.nvidia.com/isaacsim/latest/features/sensors_simulation/isaac_sim_sensors_physics_articulation_force.html)