isaac-sim/IsaacLab #103: [ ModuleNotFoundError: No module named 'omni.isaac']

zydczqcnydx, 2023-08-02:
###  ModuleNotFoundError: No module named 'omni.isaac'

Why is this error reported when there is an omni library in my conda environmen.I followed the official document configuration

zydczqcnydx, 2023-08-02:
I also found out that the path in 'setup_conda. sh' needs to be modified because the shell I am using is zsh. Thinks.

Mayankm96, 2023-08-02:
Maybe some commands aren't being checked properly for `zsh` terminal in the `orbit.sh` script. I will have to take a closer look at this for zsh and find a solution.

If you find a fix in the meantime, would be great to learn about it.

zydczqcnydx, 2023-08-02:
I have made the necessary modifications to the zsh adaptation code for "setup_conda_env. sh" and "setup_python_env. sh". You can use zsh to run the existing code as follows:

setup_conda_env. sh
`
#!/bin/zsh  
SCRIPT_DIR=$(cd "$(dirname "$0")" && pwd)  
MY_DIR=$(realpath -s "$SCRIPT_DIR")  
  
export CARB_APP_PATH=$SCRIPT_DIR/kit  
export EXP_PATH=$MY_DIR/apps  
export ISAAC_PATH=$MY_DIR  
source $ISAAC_PATH/setup_python_env.sh  
`
setup_python_env. sh
`
#!/bin/zsh  
SCRIPT_DIR=$(dirname "${(%):-%x}")  
export PYTHONPATH=$PYTHONPATH:$SCRIPT_DIR/../../../$PYTHONPATH:$SCRIPT_DIR/exts/omni.isaac.kit:$SCRIPT_DIR/exts/omni.isaac.gym:$SCRIPT_DIR/kit/kernel/py:$SCRIPT_DIR/kit/plugins/bindings-python:$SCRIPT_DIR/kit/extscore/omni.kit.pip_archive/pip_prebundle:$SCRIPT_DIR/exts/omni.isaac.core_archive/pip_prebundle:$SCRIPT_DIR/exts/omni.isaac.ml_archive/pip_prebundle:$SCRIPT_DIR/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1  
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$SCRIPT_DIR/../../../$LD_LIBRARY_PATH:$SCRIPT_DIR/.:$SCRIPT_DIR/exts/omni.usd.schema.isaac/bin:$SCRIPT_DIR/exts/omni.isaac.motion_planning/bin:$SCRIPT_DIR/kit:$SCRIPT_DIR/kit/kernel/plugins:$SCRIPT_DIR/kit/libs/iray:$SCRIPT_DIR/kit/plugins:$SCRIPT_DIR/kit/plugins/bindings-python:$SCRIPT_DIR/kit/plugins/carb_gfx:$SCRIPT_DIR/kit/plugins/rtx:$SCRIPT_DIR/kit/plugins/gpu.foundation:$SCRIPT_DIR/kit/extensions/extensions-bundled/bin  
`

zydczqcnydx, 2023-08-02:
I have made the necessary modifications to the zsh adaptation code for "setup_conda_env. sh" and "setup_python_env. sh". You can use zsh to run the existing code as follows:

##### setup_conda_env. sh
```
#!/bin/zsh  
SCRIPT_DIR=$(cd "$(dirname "$0")" && pwd)  
MY_DIR=$(realpath -s "$SCRIPT_DIR")  
  
export CARB_APP_PATH=$SCRIPT_DIR/kit  
export EXP_PATH=$MY_DIR/apps  
export ISAAC_PATH=$MY_DIR  
source $ISAAC_PATH/setup_python_env.sh  
```
##### setup_python_env. sh
```
#!/bin/zsh  
SCRIPT_DIR=$(dirname "${(%):-%x}")  
export PYTHONPATH=$PYTHONPATH:$SCRIPT_DIR/../../../$PYTHONPATH:$SCRIPT_DIR/exts/omni.isaac.kit:$SCRIPT_DIR/exts/omni.isaac.gym:$SCRIPT_DIR/kit/kernel/py:$SCRIPT_DIR/kit/plugins/bindings-python:$SCRIPT_DIR/kit/extscore/omni.kit.pip_archive/pip_prebundle:$SCRIPT_DIR/exts/omni.isaac.core_archive/pip_prebundle:$SCRIPT_DIR/exts/omni.isaac.ml_archive/pip_prebundle:$SCRIPT_DIR/extscache/omni.pip.torch-1_13_1-0.1.4+104.2.lx64/torch-1-13-1  
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$SCRIPT_DIR/../../../$LD_LIBRARY_PATH:$SCRIPT_DIR/.:$SCRIPT_DIR/exts/omni.usd.schema.isaac/bin:$SCRIPT_DIR/exts/omni.isaac.motion_planning/bin:$SCRIPT_DIR/kit:$SCRIPT_DIR/kit/kernel/plugins:$SCRIPT_DIR/kit/libs/iray:$SCRIPT_DIR/kit/plugins:$SCRIPT_DIR/kit/plugins/bindings-python:$SCRIPT_DIR/kit/plugins/carb_gfx:$SCRIPT_DIR/kit/plugins/rtx:$SCRIPT_DIR/kit/plugins/gpu.foundation:$SCRIPT_DIR/kit/extensions/extensions-bundled/bin  
```

amirhnourian, 2024-02-15:
Hello I am currently using shell and also encountering this error what should I do?:
Documents/GitHub/orbit/source/standalone/tutorials/00_sim/create_empty.py", line 20, in <module>
    from omni.isaac.orbit.app import AppLauncher
ModuleNotFoundError: No module named 'omni.isaac.orbit'

mattmazzola, 2024-05-10:
Based on the comments

> the necessary modifications to the zsh adaptation code

> using shell and also encountering this error

I also ran into this on with more explanation but maybe only a partial solution : #407 

Although the comment above may be more complete: https://github.com/NVIDIA-Omniverse/Orbit/issues/103#issuecomment-1662112177

closed 2024-10-02 by Dhoeller19