isaac-sim/IsaacLab #107: [Question] Please solve the contact force problem after the update on Aug 07, 2023

2361098148, 2023-08-09:
### Question

In legged_robot.py: 
` # TODO: contact forces -- Waiting for contact sensors in IsaacSim.`
` #   For now, use heuristics for flat terrain to say feet are in contact.`
` # air times`
And I find the isaac sim is able to support the contact force after last update onAug 07, 2023. So when can i use it in orbit, or if I want use it now, how to code?

The update [website:](https://docs.omniverse.nvidia.com/isaacsim/latest/isaac_sim_sensors_physics_based_contact.html)

Mayankm96, 2023-08-10:
You can use the [RigidContactView](https://docs.omniverse.nvidia.com/isaacsim/latest/ext_omni_isaac_core.html#create-rigidcontactview) class from Isaac Sim to get the contact forces through tensor APIs.

closed 2023-08-10 by Mayankm96

2361098148, 2023-08-12:
I have tried to follow the instruction, but anymal_c 's leg leave away from the body. I could not figure our the reason. This is the modification I tried:        
```
for foot_name, body_name in self._feet_body_name.items():
          # create rigid body view to track
          feet_body = RigidPrimView(
              prim_paths_expr=f"{prim_paths_expr}/{body_name}", name=foot_name, reset_xform_properties=False,
              track_contact_forces=True,  # enable contact forces
          )
          feet_body.initialize()
          # add to list
          self.feet_bodies[foot_name] = feet_body
```
This is the bug video:
[https://drive.google.com/file/d/1JhAIjPY9bcmU4yRYKdgEGe_b3da1vMG8/view?usp=sharing](url)

And I just add the  `track_contact_forces=True, `,and i got the result. When i delete this line, the simulation runs well, but I still could not get the contact force.

2361098148, 2023-08-12:
And I have tried both 2022.2.0 and 2022.2.1, the result is same