isaac-sim/IsaacLab #111: [Proposal] Add windows support

sambaPython24, 2023-08-25:
### Proposal

I propose to add a windows script ```.bat``` that is doing the same as ```orbit.sh```.  Please let me know if there are additional problems, that I do not see here.

### Motivation

Many people are using Windows instead of Linux.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)

sambaPython24, 2023-08-25:
Proposal (as a starting point)

```bat
@echo off

:: This is a bat file that translates the orbit.sh file.

:: Configurations
setlocal enabledelayedexpansion
set "ORBIT_PATH=%~dp0"

:: Helper functions
:extract_python_exe
set "build_path=%ORBIT_PATH%\_isaac_sim"
if not "%ISAACSIM_PATH%" == "" (
    set "build_path=%ISAACSIM_PATH%"
)

:: Checking if Conda environment variable is set
if not "%CONDA_PREFIX%" == "" (
    set "python_exe=%CONDA_PREFIX%\Scripts\python.exe"
) else (
    set "python_exe=%build_path%\python.bat"
)

:: Checking if Python executable exists
if not exist "%python_exe%" (
    echo [ERROR] No python executable found at path: %build_path%
    exit /b 1
)

echo %python_exe%
exit /b

:extract_isaacsim_exe
set "build_path=%ORBIT_PATH%\_isaac_sim"
if not "%ISAACSIM_PATH%" == "" (
    set "build_path=%ISAACSIM_PATH%"
)

:: Checking if IsaacSim executable exists
set "isaacsim_exe=%build_path%\isaac-sim.bat"

if not exist "%isaacsim_exe%" (
    echo [ERROR] No isaac-sim executable found at path: %build_path%
    exit /b 1
)

echo %isaacsim_exe%
exit /b

:install_orbit_extension
call :extract_python_exe
if exist "%~1\setup.py" (
    echo Installing module: %~1
    "%python_exe%" -m pip install --editable "%~1"
)
exit /b

:setup_conda_env
set "env_name=%~1"

:: Checking if Conda is installed
where conda >nul 2>nul
if %errorlevel% neq 0 (
    echo [ERROR] Conda could not be found. Please install conda and try again.
    exit /b 1
)

set "build_path=%ORBIT_PATH%\_isaac_sim"
[... 201 lines cut ...]
    call "%python_exe%" -m sphinx -b html -d _build\doctrees . _build\html
    cd /d "%~dp0"

    echo [INFO] To open documentation on the default browser, run:
    echo.
    echo      start "" "%ORBIT_PATH%\docs\_build\html\index.html"
    echo.

    shift
    goto :end
)

if "%~1"=="-h" (
    call :print_help
    exit /b 1
)

echo [Error] Invalid argument provided: %~1
call :print_help
exit /b 1

:end
endlocal

```

Mayankm96, 2023-08-25:
That's really cool! Thanks a lot for sharing the `.bat` file. Let me try to get hold of a Windows machine and see if this works. Would be great to have this merged in!

emondssp, 2023-08-31:
### Add Windows installation details to

Hey @Mayankm96 , I would participate in making this extension run in Windows

Could you add this to the documentation?

-----------

# Configuring the environment variables

The Windwos version was tested with Windows 11 Pro. 

! Note 
On Windows systems, you can find the path (in this case isaac_sim-2022.2.1) by either 
- ```Omniverse -> LIBRARY -> Isaac Sim (if installed) -> Isaac Sim Options (next to the LAUNCH button) -> Settings -> Install Path```
- ```Omniverse -> LIBRARY -> Isaac Sim (if installed) ->   LAUCNH -> Package Path in the Isaac Sim App Selector```

We will call the path "{ISAAC_SIM_PATH}\isaac_sim-*", with * corresponding to the Isaac Sim version. {ISAAC_SIM_PATH} could be replaced for by e.g. ```C:\Users\myuser\Documents\Nvidia```

For windows users, open the command line (cmd) set the path variable as

```bat
:: Set Isaac Sim root directory
set "ISAACSIM_PATH={ISAAC_SIM_PATH}\isaac_sim-2022.2.1"
```

```
set "ISAACSIM_PATH=D:\NVIDIA\LIBRARY\isaac_sim-2022.2.1"
```

```bat
:: Set Isaac Sim python executable
set "ISAACSIM_PYTHON_EXE=%ISAACSIM_PATH%\python.bat"
```

This path is only kept in the memory for one terminal session.

Remark: You can also set the path variable with the GUI environment as via SetEnvironmentVariable 

In order to set the path *permanently* (for any new terminal session), use 
```bat
setx ISAACSIM_PATH "{ISAAC_SIM_PATH}\isaac_sim-2022.2.0"
```
and 
```bat
setx ISAACSIM_PYTHON_EXE "%ISAACSIM_PATH%\python.bat"
```
*Restart* the terminal when using ```setx``` for the changes to take effect.

# Running the simulator

Check that the simulator runs as expected:
```bat
"%ISAACSIM_PATH%\isaac-sim.bat"
```

Check that the simulator runs from a standalone python script:

```bat
:: Check that python path is set correctly
"%ISAACSIM_PYTHON_EXE%" -c "print('Isaac Sim configuration is now complete.')"
```

```bat
:: Check that Isaac Sim can be launched from python
"%ISAACSIM_PYTHON_EXE%" "%ISAACSIM_PATH%\standalone_examples\api\omni.isaac.core\add_cubes.py"
```

emondssp, 2023-08-31:
- [Add to: Set up a symbolic link between the installed ...]

```bat
:: Enter the cloned repository
cd orbit

:: Create a symbolic link
mklink /D _isaac_sim "%ISAACSIM_PATH%"
```

Mayankm96, 2024-06-25:
Closing this issue as we have recently added Windows support to IsaacLab.

closed 2024-06-25 by Mayankm96