isaac-sim/IsaacLab #112: [Question] How to use your extension_examples for reinforcement learning?

sambaPython24, 2023-09-01:
### Question

There are good looking examples, that I find in ```Orbit/_isaac_sim/extension_examples``
and in [documentation examples](https://isaac-orbit.github.io/orbit/source/features/environments.html). 
However, in the RL tutorials, we are working with the classical Isaac Sim Examples, e.g. ```Isaac-Ant-v0```. 

How do I call your environments when using 
```
(orbit) python source/standalone/workflows/rl_games/train.py --task Isaac-Ant-v0 --headless
``
and how do I get a list of environments in [documentation examples](https://isaac-orbit.github.io/orbit/source/features/environments.html) that are already available (some of them seem to be under development)?

sambaPython24, 2023-09-06:
Does anybody have any experiences with this?

Mayankm96, 2024-02-10:
Hi,

Sorry for the late response.

We have updated our docs with the latest list of environments. Please check them out here: https://isaac-orbit.github.io/orbit/source/features/environments.html

You can also list all the environments using the following script:

```bash
./orbit.sh -p source/standalone/environments/list_envs.py
```

closed 2024-02-10 by Mayankm96