isaac-sim/IsaacLab #113: [Question] How to change the video quality?

sambaPython24, 2023-09-06:
### Question

We can record videos with ```--video```but how can we change the video quality?

sachinkundu, 2023-09-25:
Not sure if you are using the [RecordVideo](https://isaac-orbit.github.io/orbit/source/tutorials_envs/02_wrappers.html#wrapper-for-recording-videos) wrapper. It has a video resolution config setting, if that's what you mean by video quality. Haven't tried myself, YMMV.

Mayankm96, 2023-09-25:
Yes, it can be set from this parameter in the configuration object: https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/isaac_env_cfg.py#L47-L51

Mayankm96, 2024-02-10:
Closing this issue due to lack of inactivity. If the issue persists, please feel free to open it again.

closed 2024-02-10 by Mayankm96