isaac-sim/IsaacLab #114: [Question] Is this project maintained and does it have a future?

sambaPython24, 2023-09-07:
### Question

It appears to me, that github issues are not worked on, many parts of the documentation are missing and it is in general
a very frustrating experience to try to create your own environments, or to understand the general logic. Information needs to be gathered from distributed sources, documentation, videos, live-streams or the source code.

In contrast to that, we have much better organized packages that in parts base on mujoco and allow to actually focus on the algorithms.

Can we rely on this package to be maintained or was this just a short project?

sachinkundu, 2023-09-25:
@sambaPython24 This is opensource. We are supposed to make it grow and not just ask for stuff. If you find a problem then fix it!

DuaneNielsen, 2023-12-09:
Generally my experience with NVIDIA is they do reference implementations and encourage software developers to take things forward.  Don't forget, they are a hardware company first.  When using any of their code bases, don't expect a solid product roadmap or any kind of support.  They do a great job of taking PhD and Masters graduates, and allowing them to build and do cool things... but the likelihood of those types of people wanting to grind out a product is low.

Mayankm96, 2024-02-10:
Hi @sambaPython24 ,

Thank you for sharing your thoughts and concerns about our project. We genuinely appreciate your feedback and assure you that the project is actively maintained. We understand the frustration that can arise when certain aspects of documentation or issues seem unresolved and we apologize for any inconvenience this may have caused.

Improving interfaces and ensuring smooth user experiences are top priorities for us. We're continuously enhancing the framework to provide users like you with an easy-to-use and efficient toolset. Developing for a diverse community, especially in fields like robotics, requires extensive consideration and discussion, sometimes slowing progress. However, please rest assured that we are dedicated to making significant strides.

We have noted your comparison with other packages and appreciate your perspective. While we cannot promise a product roadmap similar to that of traditional software companies, we are committed to maintaining and advancing the project to meet the evolving needs of our users.

At NVIDIA, we value innovation and collaboration. While our primary focus may lie in hardware, we recognize the importance of robust software ecosystems. Hence, we endeavor to support and nurture projects like ours, empowering developers to push boundaries and create impactful solutions.

closed 2024-02-10 by Mayankm96