isaac-sim/IsaacLab #115: [Question] Cannot find the module omni.replicator

chaofiber, 2023-09-11:
### Question

While trying to run the rsl_rl with anymal-c under velocity environment, 
``` ./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Anymal-C-v0 --headless ```
I have the following error:
```
import omni.replicator.core as rep
ModuleNotFoundError: No module named 'omni.replicator'
```
Anyone knows why that happens?

b-vm, 2023-09-25:
Did you follow all the [installation instructions](https://isaac-orbit.github.io/orbit/source/setup/installation.html)?

Sounds like you didnt configure the python paths properly

chaofiber, 2023-09-25:
I can import omni.isaac.orbit perfectly well and can train some networks without loading the dynamic replicator, but only the omni.replicator.isaac has issues while loading. Can you @b-vm load the following two packages?
```
import omni.replicator.core as rep
import omni.replicator.isaac as rep_dr
```

closed 2023-09-25 by chaofiber