isaac-sim/IsaacLab #117: [Bug Report] Error in "Training an agent with SKRL on Isaac-Reach-Franka-v0"

kyuwon416, 2023-09-14:
### Describe the bug
Error in "Training an agent with SKRL on Isaac-Reach-Franka-v0"

In Orbit Documentation, I did 'Getting Started >> Running existing scripts >> Reinforcement Learning >> Training an agent with SKRL on Isaac-Reach-Franka-v0'

### Steps to reproduce
```shell
Traceback (most recent call last):
  File "source/standalone/workflows/skrl/train.py", line 52, in <module>
    from skrl.utils.model_instantiators import deterministic_model, gaussian_model, shared_model
ImportError: cannot import name 'deterministic_model' from 'skrl.utils.model_instantiators' (/home/vision/anaconda3/envs/orbit/lib/python3.7/site-packages/skrl/utils/model_instantiators/__init__.py)
2023-09-14 06:40:39 [5,416ms] [Warning] [carb.audio.context] 1 contexts were leaked
./orbit.sh: line 209: 21167 Segmentation fault      (core dumped) ${python_exe} $@
``` 

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 2022.2.1
- OS: Ubuntu 20.04
- GPU: RTX 3070 Ti
- CUDA: 11.7
- GPU Driver: 525.125.06

Toni-SM, 2023-09-15:
Hi @kyuwon416 

**skrl**-v1.0.0 introduced some breaking changes regarding to model instantiators.
Current Isaac Orbit version only works for skrl<=0.10.2.
When the PR https://github.com/NVIDIA-Omniverse/Orbit/pull/109 is approved, the latest version of **skrl** can be used in addition to its JAX support :)

Meanwhile, use the following command to install the old supported version:

```python
orbit -p -m pip install skrl==0.10.2
```

berttggg, 2023-09-15:
@Toni-SM 
Hi, I actually tried PR #109, and successfully ran the RL task. However, with exactly the same parameters and environments (same tasks), 1.0.0 and 0.10.2 their performance is quite different, in which in 0.10.2, my robot can successfully grasp the object, whereas 1.0.0 absolutely does not.

May I know if you have any clue? Is there anything I missed?

Mayankm96, 2024-07-29:
This should be fixed now with recent upgrades to skrl and Isaac Lab. Please re-open the issue if problem persists.

closed 2024-07-29 by Mayankm96