isaac-sim/IsaacLab #118: [Question] Closing an environment doesn't close replicator. Replicator already registered for given stage.

rdednl, 2023-09-15:
### Question

Not sure if this is the expected behavior so I'm posting it as a question.

A process which is doing robot learning with multiple sequential training procedures, will create an environment, do whatever it has to do, and then close the environment and launch a new environment. Unfortunately from what I've been trying, this is not currently possible in Orbit, as when doing `env.close()`, and then `gym.make()`,

Isaac will throw the following error:
 `
[19,623ms] [Error] [carb.physx.python] Replicator already registered for given stage.
`

Here is a minimum working example to reproduce the error:
```
from omni.isaac.kit import SimulationApp
headless = True
simulation_app = SimulationApp({"headless": headless})

import gym
import omni.isaac.orbit_envs
from omni.isaac.orbit_envs.utils import parse_env_cfg

def main():
    task_name = "Isaac-Ant-v0"
    cfg_direct = parse_env_cfg(task_name, use_gpu=True, num_envs=8)

    idx = 1

    while True:
        print(f"STARTING ENV {idx}")
        env = gym.make(task_name, cfg=cfg_direct, headless=headless)

        env.close()
        print(f"CLOSING ENV {idx}")
        idx += 1

if __name__ == "__main__":
    main()
```

This is the stack trace that it produces:
```
Exception has occurred: RuntimeError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
Accessed schema on invalid prim
  File "/home/daniele.reda/.local/share/ov/pkg/isaac_sim-2022.2.1/exts/omni.isaac.cloner/omni/isaac/cloner/cloner.py", line 255, in filter_collisions
    physx_scene.CreateInvertCollisionGroupFilterAttr().Set(True)
  File "/home/daniele.reda/code/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/isaac_env.py", line 157, in __init__
    physics_scene_path, "/World/collisions", prim_paths=self.envs_prim_paths, global_paths=global_prim_paths
  File "/home/daniele.reda/code/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/classic/ant/ant_env.py", line 44, in __init__
    super().__init__(isaac_cfg, headless=headless)
  File "/home/daniele.reda/code/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/gym/envs/registration.py", line 90, in make
    env = cls(**_kwargs)
  File "/home/daniele.reda/code/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/gym/envs/registration.py", line 129, in make
    env = spec.make(**kwargs)
  File "/home/daniele.reda/code/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/gym/envs/registration.py", line 235, in make
    return registry.make(id, **kwargs)
  File "/home/daniele.reda/code/Orbit/source/standalone/workflows/rl_games/test_reloading.py", line 18, in main
    env = gym.make(task_name, cfg=cfg_direct, headless=headless)
  File "/home/daniele.reda/code/Orbit/source/standalone/workflows/rl_games/test_reloading.py", line 25, in <module>
    main()
  File "/home/daniele.reda/code/Orbit/_isaac_sim/kit/python/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/daniele.reda/code/Orbit/_isaac_sim/kit/python/lib/python3.7/runpy.py", line 193, in _run_module_as_main (Current frame)
    "__main__", mod_spec)
RuntimeError: Accessed schema on invalid prim
```

rdednl, 2023-09-18:
I found that the issue is related to the fact that the `SimulationContext` in the environment is not closed properly in the environment, and it's variable `SimulationContext._sim_context_initialized` remains `True` across instances, which stops it from initializing a new stage after having it cleared.

`SimulationContext` provides a `clear_instance()` which solves the issue.

#119 is a PR that addresses this issue.

Mayankm96, 2024-02-10:
Thank you for bringing up the issue and also sending a fix for it. This has been accounted for in the latest release.

> https://github.com/NVIDIA-Omniverse/orbit/blob/main/source/extensions/omni.isaac.orbit/omni/isaac/orbit/envs/base_env.py#L303

closed 2024-02-10 by Mayankm96