isaac-sim/IsaacLab #121: [Question] Deformable Env

YunxiTang, 2023-09-23:
Dear team,
Thanks for this awesome project! I go through the related documents and tried to find some examples of deformable tasks. It seems that the deformable manipulation parts have not been included. Do you have any updated plan or timelines for the following development?
Thanks

sachinkundu, 2023-09-25:
Same question. I won't ask for timelines as you are working as fast as you can. I will ask, how can I help? Surely could write an environment myself but then don't want to duplicate the effort which you might be working on. Is there some developer call you organize?

Mayankm96, 2023-09-25:
Hi, 

I agree with @sachinkundu here. Once we are past the current release (discussion #106), we will realign our efforts and be more in sync with the community. This includes having a better roadmap and more participation from the community to help develop new features.

Deformable environment simulation slipped down in our priorities due to recent refactorings. Additionally, we have been waiting for PhysX TensorAPIs for some of the deformable object simulations. Some of these might make it into the next release of Isaac Sim, based on which we will do better feature triaging.

YunxiTang, 2023-09-26:
Hi,
Thanks for your reply and I understood the plan of your team. Looking forward to the new release with better features. I think I can close this question now.

closed 2023-09-26 by YunxiTang