isaac-sim/IsaacLab #122: [Question] How to get the position (translations) of robotic parts?

sambaPython24, 2023-09-23:
### Question

I am creating a franka robot that consist of many different parts such as ```panda_link```and ```panda_hand``.

When I try to spawn the robots
```python
  if args_cli.robot == "franka_panda":
      robot_cfg = FRANKA_PANDA_ARM_WITH_PANDA_HAND_CFG
  elif args_cli.robot == "ur10":
      robot_cfg = UR10_CFG
  else:
      raise ValueError(f"Robot {args_cli.robot} is not supported. Valid: franka_panda, ur10")
```
and 
```
    # -- Spawn robot
    robot = SingleArmManipulator(cfg = robot_cfg)
    robot_prim = "/World/envs/env_0/Robot"
    robot.spawn(robot_prim, translation = (0.0, 0.5, 0.0))
```

I just want to get the position of the robot_hand in cartesian coordinates relative to the 
world and relative to the environment. 

How do I do that?

Mayankm96, 2023-09-25:
You should be able to get that data from the `data` attribute in the robot. This data class object is filled with the buffers when you call `robot.update(dt)`.

Mayankm96, 2024-02-10:
Please check the discussion [here](https://isaac-orbit.github.io/orbit/source/tutorials/01_assets/run_rigid_object.html#resetting-the-simulation-state) on different frames.

If you want to get transformations between two frames, we recommend checking the [`FrameTransformer`](https://isaac-orbit.github.io/orbit/source/api/orbit/omni.isaac.orbit.sensors.html#frame-transformer) class.

closed 2024-02-10 by Mayankm96