isaac-sim/IsaacLab #123: [Bug Report] Error in "Training an agent with SKRL on Isaac-Reach-Franka-v0"

XInyuSong000, 2023-09-24:
### Describe the bug

Error in "Training an agent with SKRL on Isaac-Reach-Franka-v0"

In Orbit Documentation, I did 'Getting Started >> Running existing scripts >> Reinforcement Learning >> Training an agent with SKRL on Isaac-Reach-Franka-v0'

### Steps to reproduce

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 2022.2.1
- OS: Ubuntu 22.04
- GPU: RTX 4060
- CUDA: 12.0
- GPU Driver: 525.125.06

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

XInyuSong000, 2023-09-24:
...the error is:
Traceback (most recent call last):
  File "source/standalone/workflows/skrl/train.py", line 49, in <module>
    from skrl.agents.torch.ppo import PPO, PPO_DEFAULT_CONFIG
  File "/home/bajibaji/miniconda3/envs/orbit/lib/python3.7/site-packages/skrl/agents/torch/__init__.py", line 1, in <module>
    from skrl.agents.torch.base import Agent
  File "/home/bajibaji/miniconda3/envs/orbit/lib/python3.7/site-packages/skrl/agents/torch/base.py", line 4, in <module>
    import gym, gymnasium
  File "/home/bajibaji/miniconda3/envs/orbit/lib/python3.7/site-packages/gymnasium/__init__.py", line 12, in <module>
    from gymnasium.envs.registration import (
  File "/home/bajibaji/miniconda3/envs/orbit/lib/python3.7/site-packages/gymnasium/envs/__init__.py", line 382, in <module>
    load_plugin_envs()
  File "/home/bajibaji/miniconda3/envs/orbit/lib/python3.7/site-packages/gymnasium/envs/registration.py", line 565, in load_plugin_envs
    for plugin in metadata.entry_points(group=entry_point):
TypeError: entry_points() got an unexpected keyword argument 'group'

Mayankm96, 2023-09-24:
Which version of skrl did you install?

XInyuSong000, 2023-09-24:
> Which version of skrl did you install?

I attempted to use skrl-v1.0.0 and skrl-v0.10.2
however, I encountered the same error in both versions.

Mayankm96, 2023-09-25:
Is it possible to use a version lower than that? I think @Toni-SM made a PR back then, which we tested on 10.1. Probably some breaking changes have happened since.

He also opened another MR #109 that may have the fix you need.

Currently, we are working on the next version release from our internal branch so haven't gotten around to merging it in yet.

Toni-SM, 2023-09-27:
Hi @XInyuSong000 

According to the logs, the problem seems to be related to `gymnasium`.
Which version of `gymnasium` do you have installed?

XInyuSong000, 2023-09-27:
> Hi @XInyuSong000
> 
> According to the logs, the problem seems to be related to `gymnasium`. Which version of `gymnasium` do you have installed?

I have installed version 0.28.1 of gymnasium.

XInyuSong000, 2023-09-29:
> Is it possible to use a version lower than that? I think @Toni-SM made a PR back then, which we tested on 10.1. Probably some breaking changes have happened since.
>
> He also opened another MR #109 [...]

I have tested SKRL version 0.10.1, but the error persists.

Toni-SM, 2023-09-29:
Hi @XInyuSong000 

Please, check this post: https://github.com/Farama-Foundation/Gymnasium/issues/442

XInyuSong000, 2023-09-29:
> Hi @XInyuSong000
> 
> Please, check this post: [Farama-Foundation/Gymnasium#442](https://github.com/Farama-Foundation/Gymnasium/issues/442)

Hi, I have checked the post. I ran the command "pip install --upgrade importlib_metadata", but then I encountered a new error:

Traceback (most recent call last):
  File "source/standalone/workflows/skrl/train.py", line 46, in <module>
    import gym
  File "/home/bajibaji/miniconda3/envs/orbit/lib/python3.7/site-packages/gym/__init__.py", line 13, in <module>
    from gym.envs import make, spec, register
  File "/home/bajibaji/miniconda3/envs/orbit/lib/python3.7/site-packages/gym/envs/__init__.py", line 10, in <module>
    _load_env_plugins()
  File "/home/bajibaji/miniconda3/envs/orbit/lib/python3.7/site-packages/gym/envs/registration.py", line 250, in load_env_plugins
    for plugin in metadata.entry_points().get(entry_point, []):
AttributeError: 'EntryPoints' object has no attribute 'get'

Toni-SM, 2023-09-29:
Hi @XInyuSong000 

Which version of `gym` do you have installed?

XInyuSong000, 2023-09-29:
> Hi @XInyuSong000
> 
> Which version of `gym` do you have installed?

I have installed version: 0.21.0

Toni-SM, 2023-09-30:
Hi @XInyuSong000 

Mmmmmmmm, I can run Isaac Orbit's skrl examples without any problem with the following package versions:

```txt
toni@HP-ZBook-Studio-G8:~/isaacorbit/Orbit$ orbit -p -m pip show gym gymnasium importlib_metadata
[INFO] Using python from: /home/toni/.local/share/ov/pkg/isaac_sim-2022.2.1/python.sh                                                                                                                       
Name: gym
Version: 0.21.0
Summary: Gym: A universal API for reinforcement learning environments.
Home-page: https://github.com/openai/gym
Author: OpenAI
Author-email: jkterry@umd.edu
License: 
Location: /home/toni/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages
Requires: cloudpickle, importlib-metadata, numpy
Required-by: omni-isaac-orbit-envs, rl-games, skrl, stable-baselines3
---
Name: gymnasium
Version: 0.28.1
Summary: A standard API for reinforcement learning and a diverse set of reference environments (formerly Gym).
Home-page: 
Author: 
Author-email: Farama Foundation <contact@farama.org>
License: MIT License
Location: /home/toni/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages
Requires: cloudpickle, farama-notifications, importlib-metadata, jax-jumpy, numpy, typing-extensions
Required-by: skrl
---
Name: importlib-metadata
Version: 4.13.0
Summary: Read metadata from Python packages
Home-page: https://github.com/python/importlib_metadata
Author: Jason R. Coombs
Author-email: jaraco@jaraco.com
License: 
Location: /home/toni/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages
Requires: typing-extensions, zipp
Required-by: gym, gymnasium, huggingface-hub, jsonschema, Markdown, omni-isaac-orbit-envs, Pint, pluggy, pre-commit, prettytable, pytest, stable-baselines3, virtualenv
```
 
Can you please, run the examples without conda?

closed 2023-10-04 by XInyuSong000