isaac-sim/IsaacLab #128: [Bug Report] Distance joints are broken

b-vm, 2023-09-28:
### Describe the bug

Distance joints are broken when `num_envs` > 1

The below image shows the init position when `num_envs` = 2. You can see that the left robot is correct. The right robot is wrong. As soon as the sim starts running, the right robot will fly and spin uncontrolled.

This bug can be toggled in 2 ways:
- set `num_envs = 1`
- disable the distance joint in the robot

Maybe the cloner does something wrong with distance joints?

[image]

### Steps to reproduce

Run the included file `testbot.py`, which loads `testbot.usda` in a `VelocityEnv`:
[testbot.zip](https://github.com/NVIDIA-Omniverse/Orbit/files/12753612/testbot.zip)

To toggle the bug, set `num_envs = 1` or disable the distance joint in the .usda

### System Info

Describe the characteristic of your environment:

- Commit: [f2d97bdcddb3005d17d0ebd1546c7064bc7ae8bc]
- Isaac Sim Version: [2022.2.0]
- OS: [Ubuntu 20.04]
- GPU: [RTX 3060]
- CUDA: [12.0]
- GPU Driver: [525.125.06]

### Additional context

The issue does not appear when loading multiple robots in Isaac Sim, on GPU pipeline and hit play. Even with same physics params.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

b-vm, 2023-10-02:
I was able to reproduce this issue on another machine:

Commit: [2851bb33f455df40cb57e22a33627610020dfd46]
Isaac Sim Version: [2022.2.0]
OS: [Ubuntu 20.04.1]
GPU: [RTX 4090]
CUDA: [12.2]
GPU Driver: [535.104.05]

Mayankm96, 2023-10-03:
Hi @b-vm ,

Does this issue also happen when you use the CPU pipeline? You can enable CPU simulation by passing the `--cpu` argument.

We have seen some issues with floating base robots in 2022.2.1 with GPU simulation. Will need to check if this is one of them.

b-vm, 2023-10-03:
Just tried, it works on CPU! Perfectly stable. Same results on our full Digit model with 2 distance joints. 

Any chance this will be fixed in the next release?

Mayankm96, 2023-10-03:
From what I checked for ANYmal robots, the issues we were seeing in 2022.2.1 have been resolved in 2023.1. So very likely, this might be fixed as well.

If you can share the assets and a minimal script (something similar to `play_cloner.py`),  I can share it with the team, and we can check if the issue persists.

b-vm, 2023-10-03:
Sounds good!

The   [files to reproduce](https://github.com/NVIDIA-Omniverse/Orbit/files/12753612/testbot.zip) are linked in the original post. This contains the `.usda` of the bots in the screenshot and a minimal script to load them. Let me know if you that is not what you mean.