isaac-sim/IsaacLab #133: [Bug Report] ModuleNotFoundError: No module named 'omni.replicator' in skrl

XInyuSong000, 2023-10-05:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

### Steps to reproduce

I attempted to integrate a camera (50x50 RGB image) into the provided environment, Isaac-Lift-Franka-v0. I made modifications to both the ObservationManager and observation_space. I was then able to successfully run this environment using the specified command:
 ./orbit.sh -p source/standalone/environments/random_agent.py --task Isaac-Lift-Franka-v0 --num_envs 1

+----------------------------------------------------------------------------------------------------------------------------------+
|                                  Active Observation Terms in Group: 'policy' (shape: (10030,))                                   |
+-------+--------------------------------+--------------------------------+----------+-----------+----------+---------+------------+
| Index | Name                           |            Function            |  Shape   | Corruptor | Clipping | Scaling | Parameters |
+-------+--------------------------------+--------------------------------+----------+-----------+----------+---------+------------+
|   0   | arm_actions                    |          arm_actions           |   (7,)   |    None   |   None   |   1.0   |    None    |
|   1   | arm_dof_pos                    |          arm_dof_pos           |   (7,)   |    None   |   None   |   1.0   |    None    |
|   2   | img                            |              img               | (10000,) |    None   |   None   |   1.0   |    None    |
|   3   | object_desired_positions       |    object_desired_positions    |   (3,)   |    None   |   None   |   1.0   |    None    |
|   4   | object_relative_tool_positions | object_relative_tool_positions |   (3,)   |    None   |   None   |   1.0   |    None    |
|   5   | tool_actions                   |          tool_actions          |   (1,)   |    None   |   None   |   1.0   |    None    |
|   6   | tool_dof_pos_scaled            |      tool_dof_pos_scaled       |   (2,)   |    None   |   None   |   1.0   |    None    |
|   7   | tool_orientations              |       tool_orientations        |   (4,)   |    None   |   None   |   1.0   |    None    |
|   8   | tool_positions                 |         tool_positions         |   (3,)   |    None   |   None   |   1.0   |    None    |
+-------+--------------------------------+--------------------------------+----------+-----------+----------+---------+------------+

observation_space:
Box([-inf -inf -inf ... -inf -inf -inf], [inf inf inf ... inf inf inf], (10030,), float32)

Then I attempted to use the example provided in the skrl documentation. [torch_lift_franka_ppo.py](https://skrl.readthedocs.io/en/latest/_downloads/d1e7f4d58abf8eeb7b4ac11f599eac52/torch_lift_franka_ppo.py)

I encountered an error:
```
 from omni.isaac.orbit.sensors.camera.camera_cfg import PinholeCameraCfg
  File "/home/bajibaji/Desktop/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/camera/__init__.py", line 10, in <module>
    from .camera import Camera, CameraData
  File "/home/bajibaji/Desktop/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sensors/camera/camera.py", line 18, in <module>
    import omni.replicator.core as rep
ModuleNotFoundError: No module named 'omni.replicator'
[3.972s] Simulation App Shutting Down

```
However, I can successfully run torch_lift_franka_ppo.py without the camera.

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 2022.2.1
- OS: Ubuntu 22.04
- GPU: RTX 4060
- CUDA: 12.0
- GPU Driver: 525.125.06
- skrl: 1.0.0

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Toni-SM, 2023-10-05:
Hi @XInyuSong000 

Could you please share the script(s) you modified for testing?

I have included the `from omni.isaac.orbit.sensors.camera.camera_cfg import PinholeCameraCfg` statement in the Isaac Orbit `source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/manipulation/lift/lift_cfg.py` file (line 14) which seems to cause the error and run the skrl v-1.0.0 `torch_lift_franka_ppo.py` sample without any problem.

XInyuSong000, 2023-10-05:
> Hi @XInyuSong000
>
> Could you please share the script(s) you modified for testing?
>
> I have included the `from omni.isaac.orbit.sensors.camera.camera_cfg import PinholeCameraCfg` statement in the Isaac [...]

I modified lift_env.py and lift_cfg.py.

##lift_cfg.py:
```
# Copyright (c) 2022, NVIDIA CORPORATION & AFFILIATES, ETH Zurich, and University of Toronto
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

from omni.isaac.orbit.controllers.differential_inverse_kinematics import DifferentialInverseKinematicsCfg
from omni.isaac.orbit.objects import RigidObjectCfg
from omni.isaac.orbit.robots.config.franka import FRANKA_PANDA_ARM_WITH_PANDA_HAND_CFG
from omni.isaac.orbit.robots.single_arm import SingleArmManipulatorCfg
from omni.isaac.orbit.utils import configclass
from omni.isaac.orbit.utils.assets import ISAAC_NUCLEUS_DIR

from omni.isaac.orbit_envs.isaac_env_cfg import EnvCfg, IsaacEnvCfg, PhysxCfg, SimCfg, ViewerCfg
from omni.isaac.orbit.sensors.camera.camera_cfg import PinholeCameraCfg

##
# Scene settings
##

@configclass
class TableCfg:
    """Properties for the table."""

    # note: we use instanceable asset since it consumes less memory
    usd_path = f"{ISAAC_NUCLEUS_DIR}/Props/Mounts/SeattleLabTable/table_instanceable.usd"

@configclass
class ManipulationObjectCfg(RigidObjectCfg):
    """Properties for the object to manipulate in the scene."""

    meta_info = RigidObjectCfg.MetaInfoCfg(
        usd_path=f"{ISAAC_NUCLEUS_DIR}/Props/Blocks/DexCube/dex_cube_instanceable.usd",
        scale=(0.8, 0.8, 0.8),
    )
    init_state = RigidObjectCfg.InitialStateCfg(
        pos=(0.4, 0.0, 0.075), rot=(1.0, 0.0, 0.0, 0.0), lin_vel=(0.0, 0.0, 0.0), ang_vel=(0.0, 0.0, 0.0)
    )
    rigid_props = RigidObjectCfg.RigidBodyPropertiesCfg(
        solver_position_iteration_count=16,
        solver_velocity_iteration_count=1,
        max_angular_velocity=1000.0,
        max_linear_velocity=1000.0,
        max_depenetration_velocity=5.0,
        disable_gravity=False,
    )
    physics_material = RigidObjectCfg.PhysicsMaterialCfg(
        static_friction=0.5, dynamic_friction=0.5, restitution=0.0, prim_path="/World/Materials/cubeMaterial"
    )

@configclass
class GoalMarkerCfg:
    """Properties for visualization marker."""

    # usd file to import
    usd_path = f"{ISAAC_NUCLEUS_DIR}/Props/UIElements/frame_prim.usd"
    # scale of the asset at import
    scale = [0.05, 0.05, 0.05]  # x,y,z

@configclass
class FrameMarkerCfg:
[... 676 lines cut ...]
        return -torch.sum(torch.square(env.actions[:, :-1] - env.previous_actions[:, :-1]), dim=1)

    def penalizing_tool_action_l2(self, env: LiftEnv):
        """Penalize large values in action commands for the tool."""
        return -torch.square(env.actions[:, -1])

    def tracking_object_position_exp(self, env: LiftEnv, sigma: float, threshold: float):
        """Penalize tracking object position error using exp-kernel."""
        # distance of the end-effector to the object: (num_envs,)
        error = torch.sum(torch.square(env.object_des_pose_w[:, 0:3] - env.object.data.root_pos_w), dim=1)
        # rewarded if the object is lifted above the threshold
        return (env.object.data.root_pos_w[:, 2] > threshold) * torch.exp(-error / sigma)

    def tracking_object_position_tanh(self, env: LiftEnv, sigma: float, threshold: float):
        """Penalize tracking object position error using tanh-kernel."""
        # distance of the end-effector to the object: (num_envs,)
        distance = torch.norm(env.object_des_pose_w[:, 0:3] - env.object.data.root_pos_w, dim=1)
        # rewarded if the object is lifted above the threshold
        return (env.object.data.root_pos_w[:, 2] > threshold) * (1 - torch.tanh(distance / sigma))

    def lifting_object_success(self, env: LiftEnv, threshold: float):
        """Sparse reward if object is lifted successfully."""
        return torch.where(env.object.data.root_pos_w[:, 2] > threshold, 1.0, 0.0)

```

XInyuSong000, 2023-10-09:
I encountered this error through running this command: ./orbit.sh -p torch_lift_franka_ppo.py --num_envs 1 --headless

I can successfully run the demo without --headless

XInyuSong000, 2023-10-09:
> I modified lift_env.py and lift_cfg.py.
>
> ##lift_cfg.py:
>
> ```
> # Copyright (c) 2022, NVIDIA CORPORATION & AFFILIATES, ETH Zurich, and University of Toronto
> # All rights reserved.
> #
> # [...]

I encountered this error through running this command: ./orbit.sh -p torch_lift_franka_ppo.py --num_envs 1 --headless

I can successfully run the demo without --headless

Toni-SM, 2023-10-09:
Hi @XInyuSong000 

Since both modes (headless and non-headless) use different Kit experience configuration files (`.kit` files), it is necessary to enable the `omni.replicator.isaac` extension (and the `omni.kit.window.viewport`) in headless mode before importing the camera as follows:

**lift_cfg.py**
```python
from omni.isaac.core.utils.extensions import enable_extension
enable_extension("omni.replicator.isaac")
enable_extension("omni.kit.window.viewport")
from omni.isaac.orbit.sensors.camera.camera_cfg import PinholeCameraCfg
```

XInyuSong000, 2023-10-09:
> Hi @XInyuSong000
>
> Since both modes (headless and non-headless) use different Kit experience configuration files (`.kit` files), it is necessary to enable the `omni.replicator.isaac` extension (and [...]

Thank you. I've resolved the issue. However, when I attempted to run it after changing num_envs, I encountered another bug.
```
 ./orbit.sh -p torch_lift_franka_ppo.py --num_envs 2 --headless
```
```
[INFO]: Completed setting up the environment...
[skrl:INFO] Environment class: gym.core.Wrapper
[skrl:INFO] Environment wrapper: Isaac Orbit
Traceback (most recent call last):
  File "torch_lift_franka_ppo_2.py", line 116, in <module>
    trainer.train()
  File "/home/bajibaji/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/skrl/trainers/torch/sequential.py", line 77, in train
    self.single_agent_train()
  File "/home/bajibaji/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/skrl/trainers/torch/base.py", line 163, in single_agent_train
    states, infos = self.env.reset()
  File "/home/bajibaji/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/skrl/envs/wrappers/torch/isaac_orbit_envs.py", line 40, in reset
    self._obs_dict = self._env.reset()
  File "/home/bajibaji/.local/share/ov/pkg/isaac_sim-2022.2.1/kit/python/lib/python3.7/site-packages/gym/wrappers/order_enforcing.py", line 16, in reset
    return self.env.reset(**kwargs)
  File "/home/bajibaji/Desktop/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/isaac_env.py", line 229, in reset
    self._last_obs_buf = self._get_observations()
  File "/home/bajibaji/Desktop/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/manipulation/lift/lift_env.py", line 204, in _get_observations
    return self._observation_manager.compute()
  File "/home/bajibaji/Desktop/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/mdp/observation_manager.py", line 284, in compute
    self._obs_buffer[group_name] = torch.cat(list(group_obs.values()), dim=-1)
RuntimeError: Sizes of tensors must match except in dimension 1. Expected size 2 but got size 1 for tensor number 2 in the list.
[skrl:INFO] Closing environment
2023-10-09 18:58:41 [13,366ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling setWorldTransform for prim /World/envs/env_0/Robot/panda_link0 that has not been populated
[skrl:INFO] Environment closed
[13.545s] Simulation App Shutting Down

```
Did I overlook something in the environment configuration? The observation space is correct, and I can execute this demo with "--num_envs 1"

commit cf737f72 (2023-10-31): Adds in frame transformer and demo script (#183)

# Description
Adds in a new `Sensor` that performs frame transforms.

No tests as part of this initial PR as this feature is needed by
developers. Tests will be added in with the improvements:
https://github.com/isaac-orbit/orbit/issues/202.

Fixes #133 

## Type of change
- New feature (non-breaking change which adds functionality)

## Screenshots

[image]

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./orbit.sh --format`
- [x] I have made corresponding changes to the documentation
- [x] My changes ge [... cut]
closed 2023-10-09 by XInyuSong000