isaac-sim/IsaacLab #134: When do you plan to update rsl_rl

chaofiber, 2023-10-16:
Hello authors,

Thank you very much for the new release on the devel branch, this new branch has a much higher readability. I am wondering when do you plan to update the new rsl_rl package, since I believe the existing code https://github.com/leggedrobotics/rsl_rl.git is not updated and it reports some bugs due to interface changes. Alternatively, it would also be great if you can pinpoint changes that I need to make in the rsl_rl source code, it seems the changes should be minor. Thank you.

Best,

Mayankm96, 2023-10-20:
Hi,

We have made a [MR](https://github.com/leggedrobotics/rsl_rl/pull/14/) on the RSL-RL repository, which should have the necessary updates. You should be able to install the repo from the source by checking out the appropriate branch there. Let us know if you face any issues.

chaofiber, 2023-10-23:
Thanks, apart from the function in line 157 of runners/on_policy_runner.py brings up an error, the code works very well. Just commenting out that line is enough to fix the error.

Mayankm96, 2023-11-04:
We have now merged the devel branch of rsl_rl to main. You can use the release branch v2.0: https://github.com/leggedrobotics/rsl_rl/releases/tag/v2.0.0

closed 2023-11-04 by Mayankm96