isaac-sim/IsaacLab #136: [Question] Question title

ninebestwon, 2023-10-19:
### Question

[image]

An error occurred while importing the custom usd file for reinforcement learning with Orbit. Since the error code was Isaac-sim Error code, I thought it was a problem related to Isaac-sim, but I confirmed that it was being imported commonly in Isaac-sim, and the joint was also working well.

If you look at the error code below, it says, "(/World/envs/env_0/Valve/body/body) missing xformstack reset when child of rigid body (/World/envs/env_0/Valve) in hierarchy." I suspected the problem occurred between prim path "(/World/envs/env_0/Valve)" and the custom USD file in the spawning process.

Do you know if there is a solution?

### Python File
```python
class ValveEnv(IssacEnv):
    def __init__(self, cfg: ValveEnvCfg = None, **kwargs):
        ...
        self.valve = RigidObject(cfg=self.cfg.valve)
        ...

    def _design_scene(self) -> List[str]:
        ...
        self.valve.spawn(self.template_env_ns + "/Valve")
        ...
```

### Error Code
```bash
[3.037s] Simulation App Starting
[5.987s] app ready
[6.061s] RTX ready
[6.139s] Simulation App Startup Complete
[Config]: Overriding number of environments to: 1
omniverse://localhost/NVIDIA/Assets/Isaac/2022.2.1/Isaac/Samples/Orbit
[INFO]: Parsing default environment configuration from: /home/vision/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/manipulation/valve_custom/valve_cfg.py
[INFO]: Simulation pipeline:  GPU
[INFO]: Simulation device  :  GPU
Using default joint controller...
[INFO]: Isaac Orbit environment:
         Number of instances : 1
         Environment device  : cuda:0
         Physics step-size   : 0.01
         Rendering step-size : 0.01
[6.275s] [ext: omni.physx.flatcache-104.2.4-5.1] startup
2023-10-19 08:29:34 [6,309ms] [Warning] [carb.flatcache.plugin] UsdRelationship /Render/RenderProduct_omni_kit_widget_viewport_ViewportTexture_0.orderedVars has multiple targets, which is not supported
 
2023-10-19 08:29:36 [7,587ms] [Error] [omni.physicsschema.plugin] Rigid Body of (/World/envs/env_0/Valve/body/body) missing xformstack reset when child of rigid body (/World/envs/env_0/Valve) in hierarchy. Simulation of multiple RigidBodyAPI's in a hierarchy will cause unpredicted results. Please fix the hierarchy or use XformStack reset.
2023-10-19 08:29:36 [7,587ms] [Error] [omni.physicsschema.plugin] Rigid Body of (/World/envs/env_0/Valve/handle/handle) missing xformstack reset when child of rigid body (/World/envs/env_0/Valve) in hierarchy. Simulation of multiple RigidBodyAPI's in a hierarchy will cause unpredicted results. Please fix the hierarchy or use XformStack reset.
2023-10-19 08:29:36 [7,673ms] [Error] [omni.physx.plugin] PhysicsUSD: CreateJoint - no bodies defined at body0 and body1, joint prim: /World/envs/env_0/Valve/body/latch_joint**
~*~*~*
~*~*~* Direct GPU Helper:
~*~*~*   1 rigid dynamic
~*~*~*   1 articulation, maxLinks=12
~*~*~*
./orbit.sh: line 209: 99414 Segmentation fault      (core dumped) ${python_exe} $@
```
[image]

Mayankm96, 2023-10-23:
Wondering if you tested in Isaac Sim with GPU simulation as well. There is a difference in CPU and GPU simulation with Omniverse Physics so it might be the issue only comes when you do GPU simulation.

On the error message itself, I think this is happening because you have cascaded rigid prims (as the message says). It is hard for me to debug without seeing the USD asset itself.

ninebestwon, 2023-10-24:
[image]
Thank you for answering me. However, the problem only occurs when created with the official documentation provided by ORBIT: [.](https://isaac-orbit.github.io/orbit/source/tutorials_envs/01_create_env.html) The code I wrote using the ORBIT framework worked without any problems, as shown in the picture. Therefore, I suspected the problem occurred between prim path "(/World/envs/env_0/Valve)" and the custom USD file in the spawning process.