isaac-sim/IsaacLab #137: Implicit actuator with joint velocity control in Orbit/devel

chaofiber, 2023-10-23:
Hi,
I have been trying to use the implicit actuator with velocity control in the devel branch using my own robot model. I used it in the main branch and it worked pretty well. However, I saw some issues in the devel branch:
1. An bug in line 150 of file *joint_action.py*. it should be *cfg.use_default_offset*
2. When I manually set the target velocity in the *write_data_to_sim* function in *articulation.py*, the robot does not behave as I commanded. I am wondering if this function is tested given the error in point 1?

Thank you in advance.

Mayankm96, 2023-10-23:
Hi,

We haven't tested the joint action for setting velocity targets. But the demo script `play_ridgeback_franka.py` behaves the same even after the new changes. However, we had noticed earlier in 2022.2.1 that with CPU simulation it works fine, while with GPU it doesn't.

Can you please double-check that?

chaofiber, 2023-10-24:
Hi,
I switched to CPU simulation but it still doesn't work for me. However, I also tested the script ```play_ridgeback_franka.py``` and it is indeed behaving as desired. Would it be possible there exists some parsing errors due to the code refactoring? Can you test from your side by just sending a few fixed velocity command to the robot and see how it moves?

Mayankm96, 2023-10-25:
Can you share some information on what actuator models you are using?

chaofiber, 2023-11-02:
Hi,
About the second point, I was using the implicit actuator model with velocity control. However, the bug disappears when I reinstalled the isaac_sim 2022. Not sure if there is any hidden update from nvidia, but if you encounter similar issues, maybe reinstalling isaac sim could help.

Mayankm96, 2023-12-06:
Strange. Might have been the Orbit installation wasn't updated. But glad it got fixed for now. Closing the issue.

closed 2023-12-06 by Mayankm96