isaac-sim/IsaacLab #138: [Bug Report] Shape Mismatch when playing Imitation learning model.

AmanuelErgogo, 2023-10-25:
### Describe the bug

Thanks for this great work!

The issue occurs when trying to perform inference with the model trained using imitation learning in Isaac-orbit. During training, I successfully trained the model with a specific image observation shape of (480, 640, 3), but when I play the model for inference with a frame shape of (480, 640, 3), assertion error (`asset frame_input_shape == desired_frame_input_shape`) showing a shape mismatch occurs. given that shape of the frame during training and inference is the same. 

I have also tried to record and train with different shape of image but it didn't solve. 
I appreciate your help.

### Steps 
1. collected demo and trained a model
`./orbit.sh -p source/standalone/workflows/robomimic/train.py --task Isaac-Scoop-Xarm-v1 --algo bc --dataset logs/robomimic/Isaac-Scoop-Xarm-v1/hdf_dataset.hdf5`
3. Play the model
`./orbit.sh -p source/standalone/workflows/robomimic/play.py --task Isaac-Scoop-Xarm-v1 --checkpoint logs/robomimic/Isaac-Scoop-Xarm-v1/bc_img_hwc/20231024165501/models/model_epoch_200.pth`

**Error log**
```
Traceback (most recent call last):
  File "source/standalone/workflows/robomimic/play.py", line 90, in <module>
    main()
  File "source/standalone/workflows/robomimic/play.py", line 73, in main
    actions = policy(ob=obs)
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/robomimic/algo/algo.py", line 518, in __call__
    ac = self.policy.get_action(obs_dict=ob, goal_dict=goal)
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/robomimic/algo/bc.py", line 250, in get_action
    return self.nets["policy"](obs_dict, goal_dict=goal_dict)
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/robomimic/models/policy_nets.py", line 105, in forward
    actions = super(ActorNetwork, self).forward(obs=obs_dict, goal=goal_dict)["action"]
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/robomimic/models/obs_nets.py", line 584, in forward
    enc_outputs = self.nets["encoder"](**inputs)
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/robomimic/models/obs_nets.py", line 445, in forward
    self.nets[obs_group].forward(inputs[obs_group])
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/robomimic/models/obs_nets.py", line 235, in forward
    x = self.obs_nets[k](x)
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/aman/Orbit/_isaac_sim/kit/python/lib/python3.7/site-packages/robomimic/models/obs_core.py", line 172, in forward
    assert tuple(inputs.shape)[-ndim:] == tuple(self.input_shape)
AssertionError
```

### System Info

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 2022.2.0
- OS: Ubuntu 20.04
- GPU: RTX A5000
- CUDA: [e.g. 11.4]
- GPU Driver: 515.43.04

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

AmanuelErgogo, 2023-10-25:
thanks for replying.

I dropped alpha value from RGBA. 

here is the output
`input frame shape: torch.Size([480, 640, 3]), desired shape: [3, 480, 640]`

I have also tried changing shape of frame coming from simulation environment but it creates other issues

FYI, there is image pre-processor operations to process dataset / from the simulation environment (operations such as changing from uint8 to float, and re-ordering the batch to be channel-first, e.g. B-C-H-W, instead of B-H-W-C  -see [here](https://github.com/ARISE-Initiative/robomimic/blob/43d17472e64333f62c457199531a11de5a47b370/robomimic/utils/obs_utils.py#L365)
But I'm not sure how this pre-processor function is used during inference [here](https://github.com/NVIDIA-Omniverse/Orbit/blob/main/source/standalone/workflows/robomimic/play.py)

masoudmoghani, 2023-10-29:
You may want to call ObsUtils.process_obs(obs, obs_key) to get desired obs shape.

Mayankm96, 2023-12-06:
@masoudmoghani Should we add that to the play script by default? I suppose it does identity transformations if nothing is set?

AmanuelErgogo, 2023-12-06:
hi @Mayankm96 
this solved the issue for me,
I modified play script /Orbit/source/standalone/workflows/robomimic directory as follows:

       if is_rgb or is_rgbd:
            obs["front_camera_rgb"] = obs["front_camera_rgb"].permute(0, 3, 1, 2)    # change to (C, H, W) for inference
            obs["front_camera_rgb"] = obs["front_camera_rgb"].squeeze(0)                # drop dim for n_env
            assert tuple(obs["front_camera_rgb"].shape) == (3, 480, 640)

        if is_rgbd: 
            obs["front_camera_depth"] = obs["front_camera_depth"].permute(0, 3, 1, 2)
            obs["front_camera_depth"] = obs["front_camera_depth"].squeeze(0)
            assert tuple(obs["front_camera_depth"].shape) == (1, 480, 640)
        
        actions = policy(ob=obs)

closed 2024-02-10 by Mayankm96