isaac-sim/IsaacLab #142: [Question] play_* examples throw an error across multiple IsaacSim, CUDA / driver versions

kirilllzaitsev, 2023-10-28:
### Question

Hi. I'm testing out the provided examples in the `source/standalone/dmeo` directory but observing the following when trying to run `play_quadrupeds` or `play_arms`:

```
...
[INFO]: Setup complete...
>>>>>>>> Reset!
Program aborted due to an unhandled Error:
Unable to find target for this triple (no targets are registered)Fatal Python error: Aborted

Thread 0x00007f83f57fa640 (most recent call first):
  File "/opt/miniconda3/envs/orbit/lib/python3.7/concurrent/futures/thread.py", line 78 in _worker
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 870 in run
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 890 in _bootstrap
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydev_bundle/pydev_monkey.py", line 1118 in __call__

Thread 0x00007f885bfff640 (most recent call first):
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 300 in wait
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 552 in wait
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/pydevd.py", line 261 in _on_run
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_daemon_thread.py", line 49 in run
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007f8868d4a640 (most recent call first):
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 300 in wait
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 552 in wait
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/pydevd.py", line 215 in _on_run
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_daemon_thread.py", line 49 in run
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007f886954b640 (most recent call first):
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_comm.py", line 204 in _read_line
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_comm.py", line 222 in _on_run
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_daemon_thread.py", line 49 in run
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 890 in _bootstrap

Thread 0x00007f8869d4c640 (most recent call first):
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 300 in wait
  File "/opt/miniconda3/envs/orbit/lib/python3.7/queue.py", line 179 in get
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_comm.py", line 368 in _on_run
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_daemon_thread.py", line 49 in run
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 926 in _bootstrap_inner
  File "/opt/miniconda3/envs/orbit/lib/python3.7/threading.py", line 890 in _bootstrap

Current thread 0x00007f886be7e5c0 (most recent call first):
  File "/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/rigid_object/rigid_object.py", line 344 in _update_common_data
  File "/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/articulation/articulation.py", line 136 in update
  File "/orbit/source/standalone/demo/play_quadrupeds.py", line 117 in main
  File "/orbit/source/standalone/demo/play_quadrupeds.py", line 122 in <module>
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 124 in _run_code
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 136 in _run_module_code
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/_vendored/pydevd/_pydevd_bundle/pydevd_runpy.py", line 322 in run_path
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 284 in run_file
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/../debugpy/server/cli.py", line 430 in main
  File "/home/user/.vscode/extensions/ms-python.python-2023.18.0/pythonFiles/lib/python/debugpy/adapter/../../debugpy/launcher/../../debugpy/__main__.py", line 39 in <module>
  File "/opt/miniconda3/envs/orbit/lib/python3.7/runpy.py", line 85 in _run_code
  File "/opt/miniconda3/envs/orbit/lib/python3.7/runpy.py", line 193 in _run_module_as_main 
```
 The setups I tried: IsaacSim 2022.2.1 / 2023.1.0, CUDA 11.7/12.0, Nvidia driver 515/525.

Also adding the warnings before the error:

```
2023-10-28 21:01:31 [9,004ms] [Warning] [omni.isaac.core.utils.nucleus] /persistent/app/omniverse/mountedDrives setting not found
2023-10-28 21:01:31 [9,004ms] [Warning] [omni.client.plugin]  HTTP Client: provider_http: CC-493: Request through cache failed. Retrying without cache for http://omniverse-content-production.s3-us-west-2.amazonaws.com/.cloudfront.toml
[... 8 lines cut ...]
2023-10-28 21:01:35 [13,261ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /World/Anymal_b/Robot_2/LF_SHANK/collisions/mesh_0 is using obsolete 'customGeometry' attribute. To toggle custom geometry for cylinders, use the physics settings option.
2023-10-28 21:01:35 [13,261ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /World/Anymal_b/Robot_2/LH_THIGH/collisions/mesh_0 is using obsolete 'customGeometry' attribute. To toggle custom geometry for cylinders, use the physics settings option.
2023-10-28 21:01:35 [13,261ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /World/Anymal_b/Robot_2/LH_SHANK/collisions/mesh_0 is using obsolete 'customGeometry' attribute. To toggle custom geometry for cylinders, use the physics settings option.
2023-10-28 21:01:35 [13,261ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /World/Anymal_b/Robot_2/RF_THIGH/collisions/mesh_0 is using obsolete 'customGeometry' attribute. To toggle custom geometry for cylinders, use the physics settings option.
2023-10-28 21:01:35 [13,261ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /World/Anymal_b/Robot_2/RF_SHANK/collisions/mesh_0 is using obsolete 'customGeometry' attribute. To toggle custom geometry for cylinders, use the physics settings option.
2023-10-28 21:01:35 [13,261ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /World/Anymal_b/Robot_2/RH_THIGH/collisions/mesh_0 is using obsolete 'customGeometry' attribute. To toggle custom geometry for cylinders, use the physics settings option.
2023-10-28 21:01:35 [13,261ms] [Warning] [omni.physx.plugin] PhysicsUSD: Prim at path /World/Anymal_b/Robot_2/RH_SHANK/collisions/mesh_0 is using obsolete 'customGeometry' attribute. To toggle custom geometry for cylinders, use the physics settings option.
2023-10-28 21:01:37 [14,877ms] [Warning] [omni.hydra.scene_delegate.plugin] Calling getBypassRenderSkelMeshProcessing for prim /World/Anymal_c/Robot_1/base/visuals.proto_mesh_12_id12 that has not been populated
2023-10-28 21:01:37 [14,879ms] [Warning] [omni.hydra] Mesh '/World/Anymal_b/Robot_1/RH_SHANK/visuals.proto_mesh_1_id1' has corrupted data in primvar 'st_2': buffer size 117 doesn't match expected size 13812 in faceVarying primvars
[... cut ...]
2023-10-28 21:01:37 [14,892ms] [Warning] [omni.hydra] Mesh '/World/Anymal_b/Robot_1/base/visuals.proto_mesh_0_id0' has corrupted data in primvar 'st_2': buffer size 24840 doesn't match expected size 390132 in faceVarying primvars
2023-10-28 21:01:37 [14,892ms] [Warning] [omni.hydra] Mesh '/World/Anymal_b/Robot_1/base/visuals.proto_mesh_0_id0' has corrupted data in primvar 'st_1': buffer size 216 doesn't match expected size 390132 in faceVarying primvars
2023-10-28 21:01:37 [14,892ms] [Warning] [omni.hydra] Mesh '/World/Anymal_b/Robot_1/base/visuals.proto_mesh_0_id0' has corrupted data in primvar 'st': buffer size 102 doesn't match expected size 390132 in faceVarying primvars
```

Mayankm96, 2023-11-01:
Can you please specify which commit you are running?

kirilllzaitsev, 2023-11-02:
> Can you please specify which commit you are running?
> 
> Please use NVIDIA GPU driver 525/535 if possible.

The latest on main, 2851bb3. Trying the latest on devel results in the same issue.

The driver I'm using is 525.125.06 with CUDA 12.0.

Mayankm96, 2023-12-06:
Did you manage to solve this?

closed 2023-12-22 by Mayankm96

kirilllzaitsev, 2023-12-31:
The problem is solved and everything works in `2023.1.0-hotfix.1`. The problem was due to mounting part of the local IsaacSim installation to the container, overwriting some of the files under `/isaac-sim`

commit e7d43de3 (2023-11-01): Reorganizes configuration files for locomotion velocity environment (#220)

# Description

This MR does the following:

* Changes the way environment and agent configs are handled. These are
now moved to their respective `env` folder.
* Replaces YAML file with configclasses for rsl_rl library
* Adds examples of other robots for locomotion-velocity task: anymal-b,
unitree-a1, and anymal-d (placeholder)

The new environment folder structure is as follows:

```
my_task (folder)
   |- mdp  (folder where all user's task-specific terms are stored)
   |- config
       |- robot_name_1  (folder)
       [... cut]