isaac-sim/IsaacLab #143: [Question] When will the devel branch be pushed into main?

nikepupu, 2023-11-01:
### Question

I am trying to use the devel branch to run some RL, since I want to migrate some code into 2023. However, I encounter the following issues: 

./orbit.sh -p source/standalone/workflows/sb3/train.py  --task Isaac-Cartpole-v0 --num_envs 128

File "/home/nikepupu/Desktop/Orbit/source/standalone/workflows/sb3/train.py", line 137, in <module>
    main()
  File "/home/nikepupu/Desktop/Orbit/source/standalone/workflows/sb3/train.py", line 71, in main
    env_cfg = parse_env_cfg(args_cli.task, use_gpu=not args_cli.cpu, num_envs=args_cli.num_envs)
  File "/home/nikepupu/Desktop/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/utils/parse_cfg.py", line 115, in parse_env_cfg
    cfg = load_default_env_cfg(task_name)
  File "/home/nikepupu/Desktop/Orbit/source/extensions/omni.isaac.orbit_envs/omni/isaac/orbit_envs/utils/parse_cfg.py", line 39, in load_default_env_cfg
    cfg_entry_point: str = gym.spec(task_name)._kwargs.pop("cfg_entry_point")
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/gym/envs/registration.py", line 239, in spec
    return registry.spec(id)
  File "/home/nikepupu/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/python/lib/python3.10/site-packages/gym/envs/registration.py", line 203, in spec
    raise error.UnregisteredEnv("No registered env with id: {}".format(id))
gym.error.UnregisteredEnv: No registered env with id: Isaac-Cartpole-v0

It seems a lot of the files are not longer with the main branch. So what is the best way to migrant main to devel?

Mayankm96, 2023-11-02:
Thank you for your question.

Currently, we are in the process of adding numerous examples and environments back to the devel branch. However, as you noticed, not all environments have been reintegrated at this point. This is a key reason for the delay in merging the branches.

At present, you can test out the locomotion task by running one of the following:

```bash
# Headless execution
./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --headless --num_envs 4096
# Headless with off-screen render and video record:
 ./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --headless --num_envs 4096 --video --offscreen_render
# No headless execution:
 ./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task=Isaac-Velocity-Rough-Anymal-C-v0 --num_envs 4096
```
We hope to migrate the other environments within this week or next and finalize the release.

Since the last public release, we revisited the design philosophy of the framework and wanted to enhance its various functionalities. Consequently, we have had to rewrite significant portions of our codebase. This has resulted in a substantial divergence between the main and devel branches, making them largely incompatible.

To facilitate a smooth transition, we are diligently updating all the demo scripts from our previous release. This effort is aimed at allowing our users to understand the API changes and how they impact their work.

We understand that this migration may be inconvenient for our existing users. However, we believe that the new features and improvements in the devel branch will ultimately provide significant benefits, making the transition worthwhile.

evan-brooks, 2023-11-02:
Thanks for all your efforts and your progress update!

commit 3149af00 (2023-11-02): Renames Gym Envs related extensions to Tasks (#228)

# Description

Since there is a lot of confusion and debate around what is a scene vs.
environment vs. task, this MR makes it explicit that Gym/RL-related
definitions are always called "tasks". Thus, the corresponding
extensions are renamed from `orbit_envs` to `orbit_tasks`.

Fixes #143

## Type of change

- Breaking change (fix or feature that would cause existing
functionality to not work as expected)
- This change requires a documentation update

## Checklist

- [x] I have run the [`pre-commit` checks](https://pre-commit.com/) with
`./or [... cut]
closed 2023-12-22 by Mayankm96