isaac-sim/IsaacLab #146: [Question] error when installing extensions

AaboutL, 2023-11-03:
[INFO] Installing extensions inside orbit repository...                                                                                                                                                   
	 module: /home/ts/workspace/Orbit/source/extensions/omni.isaac.contrib_envs
Obtaining file:///home/ts/workspace/Orbit/source/extensions/omni.isaac.contrib_envs
  Preparing metadata (setup.py) ... done
Installing collected packages: omni-isaac-contrib-envs
  Attempting uninstall: omni-isaac-contrib-envs
    Found existing installation: omni-isaac-contrib-envs 0.0.1
    Uninstalling omni-isaac-contrib-envs-0.0.1:
      Successfully uninstalled omni-isaac-contrib-envs-0.0.1
  Running setup.py develop for omni-isaac-contrib-envs
Successfully installed omni-isaac-contrib-envs-0.0.1
	 module: /home/ts/workspace/Orbit/source/extensions/omni.isaac.orbit
Obtaining file:///home/ts/workspace/Orbit/source/extensions/omni.isaac.orbit
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle (from omni-isaac-orbit==0.3.2) (1.23.5)
Requirement already satisfied: torch in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1 (from omni-isaac-orbit==0.3.2) (2.0.1+cu118)
Requirement already satisfied: prettytable==3.3.0 in /home/ts/anaconda3/envs/orbit/lib/python3.10/site-packages (from omni-isaac-orbit==0.3.2) (3.3.0)
Requirement already satisfied: hidapi in /home/ts/anaconda3/envs/orbit/lib/python3.10/site-packages (from omni-isaac-orbit==0.3.2) (0.14.0)
Requirement already satisfied: wcwidth in /home/ts/anaconda3/envs/orbit/lib/python3.10/site-packages (from prettytable==3.3.0->omni-isaac-orbit==0.3.2) (0.2.9)
Requirement already satisfied: setuptools>=19.0 in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle (from hidapi->omni-isaac-orbit==0.3.2) (68.0.0)
Requirement already satisfied: filelock in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1 (from torch->omni-isaac-orbit==0.3.2) (3.12.2)
Requirement already satisfied: typing-extensions in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle (from torch->omni-isaac-orbit==0.3.2) (3.10.0.2)
Requirement already satisfied: sympy in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1 (from torch->omni-isaac-orbit==0.3.2) (1.12)
Requirement already satisfied: networkx in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1 (from torch->omni-isaac-orbit==0.3.2) (3.1)
Requirement already satisfied: jinja2 in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core_archive/pip_prebundle (from torch->omni-isaac-orbit==0.3.2) (3.1.2)
Requirement already satisfied: triton==2.0.0 in /home/ts/anaconda3/envs/orbit/lib/python3.10/site-packages (from torch->omni-isaac-orbit==0.3.2) (2.0.0)
Requirement already satisfied: cmake in /home/ts/anaconda3/envs/orbit/lib/python3.10/site-packages (from triton==2.0.0->torch->omni-isaac-orbit==0.3.2) (3.27.7)
Requirement already satisfied: lit in /home/ts/anaconda3/envs/orbit/lib/python3.10/site-packages (from triton==2.0.0->torch->omni-isaac-orbit==0.3.2) (17.0.4)
Requirement already satisfied: MarkupSafe>=2.0 in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/exts/omni.isaac.core_archive/pip_prebundle (from jinja2->torch->omni-isaac-orbit==0.3.2) (2.1.3)
Requirement already satisfied: mpmath>=0.19 in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1 (from sympy->torch->omni-isaac-orbit==0.3.2) (1.3.0)
Installing collected packages: omni-isaac-orbit
  Attempting uninstall: omni-isaac-orbit
    Found existing installation: omni-isaac-orbit 0.3.2
    Uninstalling omni-isaac-orbit-0.3.2:
      Successfully uninstalled omni-isaac-orbit-0.3.2
  Running setup.py develop for omni-isaac-orbit
Successfully installed omni-isaac-orbit-0.3.2
	 module: /home/ts/workspace/Orbit/source/extensions/omni.isaac.orbit_envs
Obtaining file:///home/ts/workspace/Orbit/source/extensions/omni.isaac.orbit_envs
  Preparing metadata (setup.py) ... done
Requirement already satisfied: numpy in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/kit/exts/omni.kit.pip_archive/pip_prebundle (from omni-isaac-orbit-envs==0.3.0) (1.23.5)
Requirement already satisfied: torch in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1 (from omni-isaac-orbit-envs==0.3.0) (2.0.1+cu118)
Requirement already satisfied: torchvision>=0.14.1 in /home/ts/.local/share/ov/pkg/isaac_sim-2023.1.0/extscache/omni.pip.torch-2_0_1-2.0.2+105.1.lx64/torch-2-0-1 (from omni-isaac-orbit-envs==0.3.0) (0.15.2+cu118)
Collecting protobuf==3.20.2 (from omni-isaac-orbit-envs==0.3.0)
  Using cached protobuf-3.20.2-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (1.1 MB)
Collecting gym==0.21.0 (from omni-isaac-orbit-envs==0.3.0)
  Using cached gym-0.21.0.tar.gz (1.5 MB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in gym setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
[INFO] Setting up vscode settings...

===========================================================================================
How to fix this error?

hilookas, 2023-11-04:
see: https://github.com/openai/gym/issues/3176#issuecomment-1408598628

AaboutL, 2023-11-07:
@hilookas It works for me. Thank you.

commit c3c7f6bd (2023-12-13): Adds steps for deployment on SLURM clusters (#273)

# Description

Adds two options to the `docker/container.sh` file:

- `push`: which will export the latest docker container to a singluarity
image and scp it on some compute cluster
- `job`: which will launch automatically a job on a SLURM compute
cluster

Had to make minimal changes to the `Dockerfile`, the
`docker_compose.yaml`, and `env.py` which at the moment contains the
parameters for these operations. For the job settings can be formulated
in `docker/cluster/job_submit.sh`.

Currently, this setup has been tested on ETH Zurich Euler clu [... cut]
closed 2023-11-07 by AaboutL