isaac-sim/IsaacLab #147: [Bug Report] Training Failure in skrl torch_lift_franka_ppo.py at 50-60k Time Steps

XInyuSong000, 2023-11-12:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug
Hi
I tried to use the given example of skrl of torch_lift_franka_ppo.py (https://skrl.readthedocs.io/en/latest/_downloads/d1e7f4d58abf8eeb7b4ac11f599eac52/torch_lift_franka_ppo.py) to test the training.

It runs well at the beginning. But when the time step is around 50000 or 60000, the run fails and I encountered an error:

```
ValueError: Expected parameter loc (Tensor of shape (4, 8)) of distribution Normal(loc: torch.Size([4, 8]), scale: torch.Size([4, 8])) to satisfy the constraint Real(), but found invalid values:
tensor([[nan, nan, nan, nan, nan, nan, nan, nan],
        [nan, nan, nan, nan, nan, nan, nan, nan],
        [nan, nan, nan, nan, nan, nan, nan, nan],
        [nan, nan, nan, nan, nan, nan, nan, nan]], device='cuda:0',
       grad_fn=<AddmmBackward0>)
```

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristic of your environment:

- Commit: 00af35e
- Isaac Sim Version: 2022.2.1
- OS:Ubuntu 20.04
- GPU:RTX 3060 Ti
- CUDA: 12.0
- GPU Driver: 525.147.05
- skrl version: 1.0.0

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

berttggg, 2023-11-14:
I am facing the same problem too.

XInyuSong000, 2023-11-17:
> I am facing the same problem too.

Hello, I addressed this issue by removing `min_lr` from `cfg["learning_rate_scheduler_kwargs"]` . Consequently, I was able to execute the training for 200,000 steps successfully. It appears that the NaN error may have been caused by an excessively high learning rate.

Mayankm96, 2023-12-06:
This one is a bit hard to debug or catch. usually, it has to do with two things:
1. Reset is going wrong and it is set to an invalid state where everything on GPU just explodes
2. Actions that are being set are so large that the PhysX solver breaks

Case 2 might be happening unless some clipping is handled by the policy (or softmax). Internally, in Orbit, we just do an affine transformation of actions and no saturation of input actions. Some libraries such as RL-Games do the clipping but that's handled with their wrappers.

To debug which one of them it could be, try running the training with the `--cpu` flag. That might help in identifying where the issue is coming from. When running GPU kernels from Python, it sometimes doesn't catch errors in the right places.

Mayankm96, 2024-02-29:
Haven't seen any activity here for a while. Please reopen the issue or create a new one if the problem persists.

closed 2024-02-29 by Mayankm96

Lr-2002, 2024-08-10:
I've faced the same problem and found it was caused by the env give nan feedback to the reward
``` python 
env.scene['washer'].data.joint_pos
```
is all of `nan` 
but is there any tutorial to locate the reason causing this ?

GracefulMan, 2024-08-13:
Encountered the same issue.
[image]