isaac-sim/IsaacLab #148: [Bug Report] Training Issue for Unitree A1

kolabearafk, 2023-11-13:
Hi, 

The training for quadruped locomotion using task=Isaac-Velocity-Flat-Anymal-C-v0 works beautifully but there is an issue when trying out the training for the Unitree A1 robot with task=Isaac-Velocity-Flat-Unitree-A1-v0. When looking at the result, the A1 moves a bit weird and it looks unnatural. The mean reward only reached to about 8 after 300 iterations while the mean reward of Anymal training reached to about 16. How can we make the A1 training result to be as good as the Anymal one and the locomotion gait to be more natural? Thank you!

### Steps to reproduce

Run the following command using the devel branch to train A1:
./orbit.sh -p source/standalone/workflows/rsl_rl/train.py --task=Isaac-Velocity-Flat-Unitree-A1-v0 --headless

Run the following command to see the trained result:
./orbit.sh -p source/standalone/workflows/rsl_rl/play.py --task=Isaac-Velocity-Flat-Unitree-A1-v0 --num_envs 1 

### System Info

- Commit: devel branch, commit 47e4800
- Isaac Sim Version: 2023.1.0-hotfix.1
- OS: Ubuntu 20.04.6 LTS
- GPU: 2080 TI
- CUDA: 12.2
- GPU Driver: 535.104.12

### Additional context

Please see the attached video of training result and image of training log for more information. 

[image]

https://github.com/NVIDIA-Omniverse/Orbit/assets/125360715/06384d73-8562-4633-b27d-26b99433ded1

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2023-11-13:
Hi, yes. We noticed the same. One of our colleagues has been looking into it and has potentially fixed the issue. 

While the rewards we chose are similar to that from [`legged_gym`](https://github.com/leggedrobotics/legged_gym), I think the main culprit here is that the USD for Unitree A1 on the Nucleus is not correct. It was generated from some old version of Isaac Sim's URDF importer, where some of the inertial quantities were not being loaded correctly. We generated a new one this week and managed to train a locomotion policy.

We will update the asset on Nucleus and make an MR soon with the working environment.

Mayankm96, 2023-12-19:
Hi everyone, 

We have updated the assets and the environments to support the legged locomotion training for the following robots:

* Unitree: A1, Go1, Go2
* Agility: Cassie

For Go1, we are using the actuator network provided [here](https://github.com/Improbable-AI/walk-these-ways). Hopefully, this helps people who are trying sim-to-real for the robot.

closed 2023-12-19 by Mayankm96