isaac-sim/IsaacLab #149: setuptools<=66h5py

942411526, 2023-11-23:
(orbit) yx@yx-MS-7B18:~/workspace/Orbit$ ./orbit.sh --extra 
[INFO] Installing extra requirements such as learning frameworks...            
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Obtaining file:///home/yx/workspace/Orbit/source/extensions/omni.isaac.orbit_envs
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [3 lines of output]
      error in omni-isaac-orbit_envs setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers; Expected end or semicolon (after version specifier)
          setuptools<=66h5py
                    ~~~~^
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

KyleM73, 2023-12-04:
This is the same issue as #145 which is resolved by adding a comma in line 29 of `source/extensions/omni.isaac.orbit_envs/setup.py` after `"setuptools<=66”`. This has been resolved on the devel branch.

Mayankm96, 2023-12-22:
Please use the latest main. It has migrated to Isaac Sim 2023.1 where we use Python 3.10. Most of these issues should be resolved now.

closed 2023-12-22 by Mayankm96