isaac-sim/IsaacLab #151: Error in play demo

942411526, 2023-11-24:
I run the demo in Orbit/source/standalone/demo/ the simulation is crash.

14.132s] app ready
2023-11-24 02:06:55 [14,558ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2023-11-24 02:06:55 [14,558ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[14.755s] RTX ready
[15.121s] Simulation App Startup Complete
Traceback (most recent call last):
  File "/home/yx/workspace/Orbit/source/standalone/demo/play_quadrupeds.py", line 200, in <module>
    main()
  File "/home/yx/workspace/Orbit/source/standalone/demo/play_quadrupeds.py", line 109, in main
    design_scene()
  File "/home/yx/workspace/Orbit/source/standalone/demo/play_quadrupeds.py", line 59, in design_scene
    kit_utils.create_ground_plane(
  File "/home/yx/workspace/Orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/utils/kit.py", line 129, in create_ground_plane
    prim_utils.create_prim(
  File "/home/yx/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 412, in create_prim
    prim.GetAttribute(k).Set(v)
pxr.Tf.ErrorException: 
	Error in 'pxrInternal_v0_22__pxrReserved__::UsdStage::_SetValueImpl' at line 6174 in file /buildAgent/work/ac88d7d902b57417/USD/pxr/usd/usd/stage.cpp : 'Empty typeName for </World/defaultGroundPlane/AmbientLight.intensity>'
2023-11-24 02:07:01 [20,026ms] [Warning] [carb] [Plugin: omni.spectree.delegate.plugin] Module /home/yx/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.usd_resolver/bin/libomni.spectree.delegate.plugin.so remained loaded after unload request
2023-11-24 02:07:01 [20,029ms] [Warning] [omni.stageupdate.plugin] Deprecated: direct use of IStageUpdate callbacks is deprecated. Use IStageUpdate::getStageUpdate instead.
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,030ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2023-11-24 02:07:01 [20,034ms] [Warning] [carb.audio.context] 1 contexts were leaked
2023-11-24 02:07:01 [20,056ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2023-11-24 02:07:01 [20,064ms] [Warning] [omni.core.ITypeFactory] Module /home/yx/.local/share/ov/pkg/isaac_sim-2023.1.0-hotfix.1/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.

942411526, 2023-11-24:
the version is isaac_sim-2023.1.0-hotfix.1/

xlei1o, 2023-11-24:
try to use isaac_sim-2022.2.0

942411526, 2023-11-25:
> try to use isaac_sim-2022.2.0

Thank you for your reply, it work