isaac-sim/IsaacLab #157: [Bug Report] ModuleNotFoundError: No module named 'omni.isaac.orbit_assets'

taochenshh, 2023-12-05:
### Describe the bug

I am using the latest `devel` branch.

```
ModuleNotFoundError: No module named 'omni.isaac.orbit_assets'
```

### Steps to reproduce

Run the random_angent.py script in the standalone/environments.
```
python random_agent.py 
```

Mayankm96, 2023-12-06:
Hi,

Yes. We are pushing the assets to the Nucleus server from our internal git repository. We will be making this fix in the next few days. As a workaround for now, you can comment out the code where this is being called (probably only inside the cartpole environment).

Mayankm96, 2023-12-19:
This should be fixed now on latest devel. Please feel free to open an issue if it persists.

closed 2023-12-19 by Mayankm96