isaac-sim/IsaacLab #158: [Bug Report] Possible Bug: Spawning articulations not possible during runtime

jnzhnd, 2023-12-06:
### Describe the bug
When I try to spawn an articulation in a script after I've already called sim.reset(), I get the following error message:
```
[Error] [omni.physx.plugin] PhysX error: PxArticulationReducedCoordinate::updateKinematic(): it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!, FILE /buildAgent/work/74336105e89c4a74/source/physx/src/NpArticulationReducedCoordinate.cpp, LINE 691
```
If I spawn it before the first sim.reset(), everything is fine. It could also be something in my code that I'm missing - if so, please let me know.

### Steps to reproduce
This is my spawner function:
```python
def spawn_art(asset_path, usd_cache_dir, model_name):
    art_cfg = ArticulationCfg(
        spawn=sim_utils.UrdfFileCfg(asset_path=asset_path,
                                usd_dir=os.path.join(env_cfg.usd_cache_dir, model_name),
                                usd_file_name=f"{model_name}.usd",
                                force_usd_conversion=False,
                                make_instanceable=True,
                                import_inertia_tensor=True,
                                fix_base=False,
                                merge_fixed_joints=False,
                                self_collision=False,
                                default_drive_type="none",
                                default_drive_stiffness=0,
                                default_drive_damping=0),
        init_state=ArticulationCfg.InitialStateCfg(
            pos=(0.0, 0.7, 0.2),
            rot=(-0.707, 0, 0.707, 0),
            joint_pos={
                "joint_a": 0.0,
                "joint_b": 0.0,
            },
        ),
        actuators={
            "joints": ImplicitActuatorCfg(
                joint_names_expr=["joint_a",
                                "joint_b"],
                velocity_limit=0.0,
                effort_limit=0.0,
                stiffness=0.0,
                damping=0.0,
            )
        }
    )

    return Articulation(cfg=art_cfg.replace(prim_path="/World/envs/env_.*/Object"))
```
If I do the following, everything works (this is the main script):
```python
def main():
    ...
    cfg = sim_utils.SimulationCfg(dt=0.01, use_gpu_pipeline=True, device="cuda:0")
    sim = sim_utils.SimulationContext(cfg)
    ...
    my_object = spawn_art(asset_path, usd_cache_dir, model_name)
    ...
    sim.reset()

    while simulation_app.is_running():
        ...
        sim.step()
```
But if I do as follows (let's say I only want to spawn the object based on a condition happening inside the simulation loop), I get the error as described above.
```python
def main():
    ...
    cfg = sim_utils.SimulationCfg(dt=0.01, use_gpu_pipeline=True, device="cuda:0")
    sim = sim_utils.SimulationContext(cfg)
    ...
    sim.reset()

    while simulation_app.is_running():
        ...
        if condition:
            my_object = spawn_art(asset_path, usd_cache_dir, model_name)
        sim.step()
```
### System Info

- Using the devel branch of Orbit
- Commit: [aaab27b](https://github.com/NVIDIA-Omniverse/Orbit/pull/129/commits/aaab27b76a8b849f39ecd383962836da9459f344)
- Isaac Sim Version: 2023.1.0-hotfix.1
- OS: Ubuntu 22.04
- GPU: RTX 3090
- CUDA: 12.0
- GPU Driver: 525.147.05

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

Mayankm96, 2023-12-06:
Hi, dynamic spawning of objects is not recommended with the framework. 

All the scene designing must happen before the simulation starts. Once the simulation starts, we recommend keeping
the scene frozen and only altering the properties of the prim. This is particularly important for GPU simulation
as adding new prims during simulation may alter the physics simulation buffers on GPU and lead to unexpected
behaviors.

However, if you do this with CPU simulation it should not lead to errors that you see above. Though to register the physics handles for this new object in Orbit, you should then again do `sim.pause()` and `sim.play`.

jnzhnd, 2023-12-06:
Not quite what I was hoping for, but a very valid reason. Thank you very much for the quick response and the great work you're doing.

closed 2023-12-06 by jnzhnd