isaac-sim/IsaacLab #161: PhysX error: Illegal BroadPhaseUpdateData

Jianlinlynn, 2023-12-07:
### Question
Hi, I am trying to run a self-designed robot with Orbit Framewok, and I keep having the following questions:

[image]

And shown in the figure, it says, Invalid PhysX transform detected for "/World/Robot_1/logRobotDual/F_L_Lifting", which is every coordinate in the USD, as shown in the following figure:
[image]

I have no idea which part is wrong and many thanks to any advise and help!

Mayankm96, 2023-12-10:
Hi, can you please provide more information on what you are executing? These errors normally happen if some transform is being set explicitly on an articulation. Articulations in PhysX only allow setting their state using root and joint states.

Jianlinlynn, 2023-12-11:
Hi @Mayankm96, Thank you so much for your reply! I am trying to import a self-designed robot into the environment based on orbit, for now there are still no actions applied to it. Still have no idea what is going on.

Dhoeller19, 2024-10-02:
Hi @JianlinWang-JL, seems like the set-up of your articulation in USD is wrong, which leads to physx exploding.